arendst/Sonoff-MQTT-OTA-Arduino #166: Send state to sonoff

okosoko, 2016-12-28:
Hello.
I am newbe.
My question is : how send change state to sonoff from domoticz.
From sonoff to domoticz is ok, becouse state in domoticz is change when i push button.

Thanks

arendst, 2016-12-28:
If you receive status from sonoff using Domoticz MQTT then you should be able to just use the switch configured in Domoticz too. I suppose you configured the correct idx value in sonoff either via the webpage or via command DomoticzIdx1.

Wildvuur, 2016-12-29:
This is someting i did not find, i wrote a flow in node-red to parse domoticz/out to Sonoff mqtt commands.
so if someone knows how this works, a wiki addon would be nice.

Matthieu

arendst, 2016-12-29:
Matthieu et al,

I've updated the wiki with a Domoticz page explaining the configuration steps. No need for Node-Red.

Wildvuur, 2016-12-30:
Hi arendst,
the wiki is what i did before but its not working.
it looks like Sonoff wont parse the domotics/out
here is what my domoticz  sends to domotics/out for on:
`{ "topic": "domoticz/out", "payload": "{\n \"Battery\" : 255,\n \"RSSI\" : 12,\n \"description\" : \"\",\n \"dtype\" : \"Light/Switch\",\n \"id\" : \"000140EC\",\n \"idx\" : 156,\n \"name\" : \"Kerstboom\",\n \"nvalue\" : 1,\n \"stype\" : \"Switch\",\n \"svalue1\" : \"0\",\n \"switchType\" : \"On/Off\",\n \"unit\" : 1\n}\n", "qos": 0, "retain": false, "_msgid": "6f0ea87c.90f158" }

here is what my domoticz  sends to domotics/out for off:
{ "topic": "domoticz/out", "payload": "{\n \"Battery\" : 255,\n \"RSSI\" : 12,\n \"description\" : \"\",\n \"dtype\" : \"Light/Switch\",\n \"id\" : \"000140EC\",\n \"idx\" : 156,\n \"name\" : \"Kerstboom\",\n \"nvalue\" : 0,\n \"stype\" : \"Switch\",\n \"svalue1\" : \"0\",\n \"switchType\" : \"On/Off\",\n \"unit\" : 1\n}\n", "qos": 0, "retain": false, "_msgid": "7b86b3f8.84794c" }`

not sure whats wrong here, if i post {"idx":156, "nvalue":0, "svalue":""} in domoticz out it works but what domoticz sends is alot more info

thanks in advance

Matthieu

arendst, 2016-12-31:
Your domoticz payload is way too long and the important fields won't fit in sonoffs receive buffer. I've tested domoticz only with the official releases where 3.5877 is the latest.

I  do not receive description field for instance:

01:06:19 MQTT: Receive topic domoticz/out, data size 236, data {
   "Battery" : 255,
   "RSSI" : 12,
   "dtype" : "Light/Switch",
   "id" : "000140D4",
   "idx" : 142,
   "name" : "mqtt wkaku8",
   "nvalue" : 1,
   "stype" : "Switch",
   "svalue1"

Set logging to 4 and see what the received message does look like to sonoff. The important fields are idx and nvalue.

Try to fiddle around with domoticz MQTT client settings to find a shorter message. I hate to extent my receive buffer as domoticz is sending a lot of crap already...

Wildvuur, 2016-12-31:
im using beta 3.6185 
ill try this tomorow as its time to go party and so on, thanks for the fast replys!

matthieu

gorstj, 2017-01-08:
I am seeing similar behaviour, using 3.5877, Domoticz changes it's status to button presses on the Sonoff. However when I switch the 'light' on and off in Domoticz nothing happens to the Sonoff.

Here is my output from mosquitto_sub -v -t domoticz/out

domoticz/out {
   "Battery" : 255,
   "RSSI" : 12,
   "dtype" : "Lighting 1",
   "id" : "65",
   "idx" : 165,
   "name" : "Garden",
   "nvalue" : 0,
   "stype" : "X10",
   "switchType" : "On/Off",
   "unit" : 2
}

There is nothing in the Sonoff log when set to log level 4 when these messages are broadcast. 
Out topic is set as the default domoticz/out
In the MQTT settings of Sonoff I have changed hostname to 'pi' and left the port as 1883
It won't seem to let me change the client/userid/password from the defaults....but it is updating Domoticz without me changing these.

I hope someone can help - very new to this MQTT stuff

arendst, 2017-01-08:
Although I initially expected the buffer size to be a problem that is not the case. The buffer is assigned to the length of the message so the complete Domoticz message is received for decoding. 

If you enable logging with option 4 and execute the mosquitto command in your last line do you see anything in the sonoff log?

gorstj, 2017-01-08:
My fault completely!! Thanks for the gentle nudge!
I did not note the very obvious prerequesite  #define MQTT_MAX_PACKET_SIZE 400
After uploading the new code all works..... I hope this helps someone else!

Sorry to take you time!

closed 2017-01-09 by arendst

Wildvuur, 2017-01-12:
After reading up i might also made this mistake, i uploaded the latest version through the webportal, and now it works like a charm.
Thanks for this

matthieu