arendst/Sonoff-MQTT-OTA-Arduino #167: Amazon Echo plea for help

IrregularShed, 2016-12-29:
Hi folks.

I'm having issues getting started with Echo integration. I've read around other issues and the wiki to try and find the answer myself but failed completely.

I've got a couple of Sonoff devices that I've flashed with this firmware; initially with 3.0.9, but now one of them is on 3.1.5, updated OTA. The web interface is working perfectly, and I did (for a while) have my Echo Dot controlling one of them perfectly. It _did_ take a few tries to get it running, but it seemed to be that as long as I'd just accessed the `setup.xml` URL the Echo would see it when scanning.

However...
* the second one was never found by the Echo, ever
* the first one can't be found following a change to the DHCP server on my router (which was part of me trying to get the second one to be detected). I've changed the settings back (and reset/renewed leases on all relevant devices) but it remains MIA.

I've set the debug level on the web interface to `debug` but nothing is showing up in the console when the Echo is trying to detect devices, and I'm unsure what I should try next. Can anyone provide a suggestion of what to try next?

arendst, 2016-12-29:
I suppose you enabled Echo support in user_config.h (line 101) and see messages like UPnP: Multicast (re)joined

If debug is enabled and the correct M-SEARCH message is send by Echo you should see a message regarding a response sent by Sonoff.

IrregularShed, 2016-12-29:
Yep, Echo support is configured:
```
00:00:05 HTTP: Webserver active on porchlights with IP address 172.16.16.26
00:00:05 UPnP: Multicast (re)joined
```
I've also compiled it with the packetBuffer logging in support.ino. I'm not seeing anything come through when attempting to discover devices though, and looking again at the code I probably should be seeing all UPnP messages. I'll check the router configuration and see if there's something blocking things, I think.

IrregularShed, 2016-12-29:
OK, I've been sniffing with Wireshark:
[image]
172.16.16.21 is my Echo Dot, and here it's searching for "Wemo" devices. Over on the Sonoff console there's nothing being shown relating to the scan.

I'm wondering if my choice of IP ranges might be a problem. The router is on 172.16.0.1 but my DHCP pool is 172,16,16.*. The netmask is 255.255.0.0. so it shouldn't be an issue but I really don't know enough about networking (yet) to figure out where the failure is.

IrregularShed, 2016-12-30:
Right, problem solved. The issue is to do with the Echo/Alexa UI and not this (splendid) firmware. The lack of any meaningful wi-fi settings display meant that double-checking the saved configuration was impossible. Resetting the Echo's wi-fi settings had previously failed due to my phone's reaction to being connected to a network with no internet access; going back through the procedure with mobile data switched off meant that the wi-fi finally got reset (and the settings I had to change propagated through to it).

Obviously all of this could've been avoided if Amazon had included an Advanced mode (or similar) so that you could check and adjust settings without talking to the Echo, but there we go. Thanks @arendst for a great project 🙋

closed 2016-12-30 by IrregularShed

alphalove, 2017-01-04:
Hello,
Just thought I'd add some advise here if people are having issues with Alexa finding the Sonoff.

Make sure the Alexa device is connected to the same Wi-Fi network.  My router has both 5G and 2.4G Wi-Fi networks.  When the Echo is connected to 5G it can't discover the Sonoff's, when connected to the 2.4G it can (which is the same as the Sonoff's as they only support 2.4G).  I thought that these two Wi-Fi networks were essentially the same thing, but it must affect the discovery / broadcast messages.

Cheers,

alex

gadjet, 2017-01-04:
Thanks for the info.
I have different names for my 2.4GHz and 5GHz networks, this makes it easier.

Cheers,
Phil