arendst/Sonoff-MQTT-OTA-Arduino #168: Send Power State in mqtt Connect

eifinger, 2016-12-29:
Hi, could you implement that on mqtt Connect the sonoff will senden its current Power State?

This ist useful when used in a Setup where the sonoff is used after a wallswitch. We already habe the Option that ist will Turn on in bootup now I need ist to send a State Message in mqtt Connect so my homeassistant will recognize that ist came Back online.
Thank you!

arendst, 2016-12-29:
As far as I know this is already the case; as soon as Sonoff boots up, wifi and MQTT connections have been made a status message is send containing the switch state in both JSON and Legacy messageformat.

eifinger, 2016-12-29:
You are completely right.
My HA misbehaved and I couldn't find the part in the sourcecode, I just got back home and checked with mosquitto_sub.
Thanks! =)

closed 2016-12-29 by eifinger