arendst/Sonoff-MQTT-OTA-Arduino #169: Switching both relays on a Sonoff Dual results in broken state

phreakmonkey, 2016-12-29:
Sending "cmnd/Topic/POWER1 on" and "cmnd/Topic/POWER2" on in a short time period will result in the Sonoff device reporting POWERn ON for both relays, but only toggling the first one.

E.g.:  for a Sonoff Dual with topic "FrontExterior", sending:
```
$ mosquitto_pub -h mqtt -t cmnd/FrontExterior/POWER2 -m on
$ mosquitto_pub -h mqtt -t cmnd/FrontExterior/POWER1 -m on
```

results in this status on the FrontExterior sonoff console:
```
10:23:49 MQTT: stat/FrontExterior/POWER2 = ON
10:23:49 MQTT: stat/FrontExterior/POWER1 = ON
```

But only relay 2 is observed to be ON.  I also observed the "stat/" and "tele/" messages afterwards indicating "POWER1 ON" while the relay was actually OFF.   Interestingly, sending any additional POWER messages to the cmnd/FrontExterior  seems to resolve the state.

E.g. if I send the two commands to get it into this broken state with POWER1 = ON and power1 relay actually off, sending cmnd/FrontExterior/POWER2 = OFF will actually also turn POWER1 relay ON, thus resolving the inconsistent state.

I've tried this with all three QoS levels, and with two different MQTT clients (mosquito_pub and python-paho) with the same results.

Sonoff Dual firmware info:
```
Program version	3.1.4
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	49 Hours
Flash write count	83
Boot count	47
Reset reason	Software/System restart
 
AP1 SSId (RSSI)	Rainbows (48%)
Hostname	FrontExterior-2773
IP address	172.31.1.108
Gateway	172.31.1.1
MAC address	60:01:94:07:0A:D5
 
MQTT Host	mqtt.tahoe.solace.net
MQTT Port	1883
MQTT Client and
 Fallback Topic	DVES_070AD5
MQTT User	Sonoff
MQTT Topic	FrontExterior
MQTT Group Topic	sonoffs
 
ESP Chip id	461525
Flash Chip id	1327328
Flash size	1024kB
Program flash size	1024kB
Program size	380kB
Free program space	556kB
Free memory	27kB
```

arendst, 2016-12-29:
As dual uses the serial interface to control both relays did you disable seriallogging? (seriallog 0)

The problem is not MQTT related but is caused by serial queueing .

phreakmonkey, 2016-12-30:
ah, setting "seriallog 0" does indeed make the bug go away.

closed 2016-12-30 by arendst