arendst/Sonoff-MQTT-OTA-Arduino #172: Flash improvement for Sonoff RF

gadjet, 2016-12-30:
Hi,
To solve the issue of the Sonoff RF devices not paring with the RF remote I added a two pin header to the R21 position so I can add/remove a shorting plug.
[image]

arendst, 2016-12-30:
So to make it clear: 
- Add plug when you want to flash new firmware
- Remove plug when you want to pair a RF remote

Normal operation is without the plug

gadjet, 2016-12-30:
Yes that's correct.

kristoffernolgren, 2017-06-23:
I'm having a hard time getting this to work. I'm fairly certain I'm using a sonoff RF, it has the extra component with the antenna and I have tried booting it with rx-tx/tx-rx and rx-rx/rx-rx. How can you tell if the sonoff is booted in programming mode? 

When it boots normally, the relae flips once and the light starts blinking slowly and the a green amber.

When I boot it while only hodling down the button, the green light blinks quickly and the sonoff creates a new wifi network.