arendst/Sonoff-MQTT-OTA-Arduino #173: Mqtt connecting problem

Karl-opec, 2016-12-30:
I have compiled the software for sonoff_pwr. It connects fine to my wifi, but has Problems to send to the MQTT server i run. SONOFF is in subnet 192.168.40.0/24  MQTT Server in subnet 192.168.20.0/24. Sending messages to the server from different clients do work. Server is mosquitto 1.4.10.

In the Console Log of sonoff_pwr i get

00:26:36 Wifi: Checking connection...
00:26:36 Wifi: Connected
00:26:40 MQTT: Attempting connection...
00:26:46 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds

status 0 in console gives 👍 
00:28:22 HTTP: Handle console
00:28:22 MQTT: Receive topic cmnd/sonoffpow/status, data size 1, data 0
00:28:22 MQTT: DataCb Topic sonoffpow, Group 0, Index 1, Type STATUS, Data 0 (0)
00:28:22 RSLT: stat/sonoffpow/RESULT = 3.1.5, 1, sonoffpow, 0, LIGHT, 0, 3, 1, 1, 1, 1, 0, 0
00:28:22 RSLT: stat/sonoffpow/RESULT = PWR: Voltage 0 V, Current 0.000 A, Power 0 W, Today 0.000 kWh, Factor 0.00
00:28:22 RSLT: stat/sonoffpow/RESULT = PWR Threshold: PowerLow 0 W, PowerHigh 0 W, VoltageLow 0 V, VoltageHigh 0 V, CurrentLow 0 mA, CurrentHigh 0 mA
00:28:22 RSLT: stat/sonoffpow/RESULT = PRM: Baudrate 115200, GroupTopic pows, OtaUrl http://domus1:80/api/arduino/sonoffpow.ino.bin, Uptime 0 Hr, BootCount 27, SaveCount 58
00:28:22 RSLT: stat/sonoffpow/RESULT = FWR: Program 3.1.5, Boot 31, SDK 1.5.3(aec24ac9)
00:28:22 RSLT: stat/sonoffpow/RESULT = LOG: Seriallog 4, Weblog 4, Syslog 0, LogHost domus1, SSId1 xxxxx, Password1 xxxxx, SSId2 indebuurt2, Password2 VnsqrtnrsddbrN, TelePeriod 300
00:28:22 RSLT: stat/sonoffpow/RESULT = MEM: ProgramSize 388kB, Free 548kB (Heap 26kB), SpiffsStart 940kB (64kB), FlashSize 4096kB (1024kB)
00:28:22 RSLT: stat/sonoffpow/RESULT = NET: Host sonoffpow-7303, IP 192.168.40.167, Gateway 192.168.40.10, Subnetmask 255.255.255.0, Mac 5C:CF:7F:20:DC:87, Webserver 2, WifiConfig 3
00:28:22 RSLT: stat/sonoffpow/RESULT = MQT: Host odroid64, Port 1883, Client DVES_%06X (DVES_20DC87), User , Password , MAX_PACKET_SIZE 400, KEEPALIVE 15
00:28:22 RSLT: stat/sonoffpow/RESULT = TIM: UTC Thu Jan 01 00:28:22 1970, Local Thu Jan 01 00:28:22 1970, StartDST Thu Jan 01 00:00:00 1970, EndDST Thu Jan 01 00:00:00 1970
00

It also does not get the time from my ntp server.

Any hint how to debug this a bit further?
thanks

arendst, 2016-12-30:
Do you have a DNS  server for resolving your MQTT server name?
Same for your ntp servers.

Karl-opec, 2016-12-31:
of course is the MQTT and NTP Server Name resolvable. But even
with the right IP's the result is the same.

```00:03:41 MQTT: Attempting connection...
00:03:47 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds
00:03:47 HTTP: Handle console
00:03:47 MQTT: Receive topic cmnd/sonoffpow/status, data size 1, data 0
00:03:47 MQTT: DataCb Topic sonoffpow, Group 0, Index 1, Type STATUS, Data 0 (0)
00:03:47 RSLT: stat/sonoffpow/RESULT = 3.1.5, 1, sonoffpow, 0, LIGHT, 0, 3, 1, 1, 1, 1, 0, 0
00:03:47 RSLT: stat/sonoffpow/RESULT = PWR: Voltage 0 V, Current 0.000 A, Power 0 W, Today 0.000 kWh, Factor 0.00
00:03:47 RSLT: stat/sonoffpow/RESULT = PWR Threshold: PowerLow 0 W, PowerHigh 0 W, VoltageLow 0 V, VoltageHigh 0 V, CurrentLow 0 mA, CurrentHigh 0 mA
00:03:47 RSLT: stat/sonoffpow/RESULT = PRM: Baudrate 115200, GroupTopic pows, OtaUrl http://domus1:80/api/arduino/sonoffpow.ino.bin, Uptime 1 Hr, BootCount 29, SaveCount 61
00:03:47 RSLT: stat/sonoffpow/RESULT = FWR: Program 3.1.5, Boot 31, SDK 1.5.3(aec24ac9)
00:03:47 RSLT: stat/sonoffpow/RESULT = LOG: Seriallog 4, Weblog 4, Syslog 0, LogHost domus1, SSId1 giot12, Password1 32gio$*t12, SSId2 indebuurt2, Password2 VnsqrtnrsddbrN, TelePeriod 300
00:03:47 RSLT: stat/sonoffpow/RESULT = MEM: ProgramSize 388kB, Free 548kB (Heap 28kB), SpiffsStart 940kB (64kB), FlashSize 4096kB (1024kB)
00:03:47 RSLT: stat/sonoffpow/RESULT = NET: Host sonoffpow-7303, IP 192.168.40.167, Gateway 192.168.40.10, Subnetmask 255.255.255.0, Mac 5C:CF:7F:20:DC:87, Webserver 2, WifiConfig 3
00:03:47 RSLT: stat/sonoffpow/RESULT = MQT: Host 192.168.20.159, Port 1883, Client DVES_%06X (DVES_20DC87), User , Password , MAX_PACKET_SIZE 400, KEEPALIVE 15
00:03:47 RSLT: stat/sonoffpow/RESULT = TIM: UTC Thu Jan 01 00:03:47 1970, Local Thu Jan 01 00:03:47 1970, StartDST Thu Jan 01 00:00:00 1970, EndDST Thu Jan 01 00:00:00 1970

arendst, 2016-12-31:
Try entering any mqtt user and password.

Karl-opec, 2016-12-31:
Thanks for your support.
It was a wrong firewall rule dropping only new states from subnet 192.168.40.0/24.  So accessing
from subnet 192.168.20.0 worked but not vice versa.

´´´12:58:48 MQTT: Receive topic cmnd/sonoffpow/status, data size 1, data 0
12:58:48 MQTT: DataCb Topic sonoffpow, Group 0, Index 1, Type STATUS, Data 0 (0)
12:58:48 MQTT: stat/sonoffpow/RESULT = 3.1.5, 1, sonoffpow, 0, LIGHT, 0, 3, 1, 1, 1, 1, 0, 0
12:58:48 MQTT: stat/sonoffpow/RESULT = PWR: Voltage 0 V, Current 0.000 A, Power 0 W, Today 0.000 kWh, Factor 0.00
12:58:48 RSLT: stat/sonoffpow/RESULT = PWR Threshold: PowerLow 0 W, PowerHigh 0 W, VoltageLow 0 V, VoltageHigh 0 V, CurrentLow 0 mA, CurrentHigh 0 mA
12:58:48 RSLT: stat/sonoffpow/RESULT = PRM: Baudrate 115200, GroupTopic pows, OtaUrl http://domus1:80/api/arduino/sonoffpow.ino.bin, Uptime 0 Hr, BootCount 44, SaveCount 98
12:58:48 MQTT: stat/sonoffpow/RESULT = FWR: Program 3.1.5, Boot 31, SDK 1.5.3(aec24ac9)
12:58:48 RSLT: stat/sonoffpow/RESULT = LOG: Seriallog 4, Weblog 4, Syslog 0, LogHost domus1, SSId1 xxxxx, Password1 xxxxx, SSId2 indebuurt2, Password2 VnsqrtnrsddbrN, TelePeriod 300
12:58:48 MQTT: stat/sonoffpow/RESULT = MEM: ProgramSize 388kB, Free 548kB (Heap 26kB), SpiffsStart 940kB (64kB), FlashSize 4096kB (1024kB)
12:58:48 RSLT: stat/sonoffpow/RESULT = NET: Host sonoffpow-7303, IP 192.168.40.167, Gateway 192.168.40.10, Subnetmask 255.255.255.0, Mac 5C:CF:7F:20:DC:87, Webserver 2, WifiConfig 3
12:58:48 RSLT: stat/sonoffpow/RESULT = MQT: Host 192.168.20.159, Port 1883, Client DVES_%06X (DVES_20DC87), User , Password , MAX_PACKET_SIZE 400, KEEPALIVE 15
12:58:48 RSLT: stat/sonoffpow/RESULT = TIM: UTC Sat Dec 31 11:58:48 2016, Local Sat Dec 31 12:58:48 2016, StartDST Sun Mar 27 02:00:00 2016, EndDST Sun Oct 30 03:00:00 2016

Issue is solved and closed

closed 2016-12-31 by Karl-opec