arendst/Sonoff-MQTT-OTA-Arduino #174: Temperature & Humidity readings in Domoticz

rlust, 2016-12-30:
Looking for how to display the temperature & humidity readings in Domoticz?

nigelb135, 2017-01-05:
I had a similar requirement. Solved it by a Node-Red flow which 

* subscribes to the (json formatted) ``tele/+/TELEMETRY`` topic, 
* filters to ignore messages without ``DHT`` values in the payload, 
* mapping topic to the idx of the virtual sensor,
*  injecting the idx, nvalue and svalue values into the payload, 
* and rewriting the topic before,
*  republishing.

Has been working with both hacked standard sonoffs with home-made sensors and official sonoff TH10 with itead sensors since v3.1.5

Happy to share flow if anyone interested - although at the rate that Theo is pumping out new features I wouldn't be surprised if we see built in support in the future. 

For the record, a temp/hum message is published to domoticz/in with the message payload e.g. ``"idx": 5, "nvalue": 0, "svalue": "20.9;76.9;0"``. The format of the svalue is ``"temp;hum;units"`` where units is 0 for Celsius or 1 for Fahrenheit. Any extraneous content in the payload is simply ignored by domoticz.

lborek, 2017-01-05:
@nigelb135 any chance you can share the flow (with short HOWTO if possible)?

Thanks!

rlust, 2017-01-08:
Thanks nigelb135 fore your suggestion! If you could share your flow that would be very helpful.

arendst, 2017-01-27:
NextGen Version 3.9.3 contains a lot of Domoticz updates like defining sensor indexes to supported sensors. See Domoticz Configuration webpage for more information.

closed 2017-01-27 by arendst