arendst/Sonoff-MQTT-OTA-Arduino #175: Feature request: setting power state via web interface

mitkodotcom, 2016-12-30:
Hello,

Would you consider adding an option to directly set the power state via the web interface, instead of just toggle? This will be very handy for automation via IFTTT's maker channel, for example.

I made a quick patch:

    if (Maxdevice) {
      if (strlen(webServer->arg("o").c_str())) {
        if (strlen(webServer->arg("s").c_str())) {
          do_cmnd_power(atoi(webServer->arg("o").c_str()), atoi(webServer->arg("s").c_str()));
        } else
        do_cmnd_power(atoi(webServer->arg("o").c_str()), 2);
      }

This way I can set the state to on or off by adding &s=1 and &s=0 respectively, but you may think for a more elegant solution.

arendst, 2016-12-30:
If webserver 2 is active try http://sonoff/cm?cmnd=power%20on

You'll find that all commands are available this way.

mitkodotcom, 2016-12-31:
Yes, the console mode will do the trick. Thanks.

closed 2017-01-01 by arendst