arendst/Sonoff-MQTT-OTA-Arduino #176: Feature request: running without MQTT server

mitkodotcom, 2016-12-30:
If there is no MQTT server running on the configured host, the application stops responding for about 10 seconds every 30 second or so. It hangs in mqtt_reconnect() until mqttClient.connect times out. During this time the web server does not accept new connections.

arendst, 2016-12-30:
MQTT is the core of this software.

mitkodotcom, 2016-12-31:
Yes, and that's great, but it is more convenient to use one firmware for all purposes.  It's quite easy and harmless to add an "if there is no server configured, do not try to connect" statement.

Meanwhile I found another more elegant solution to my "hangs for 15 seconds every 30 seconds" problem. I have just decreased MQTT_SOCKET_TIMEOUT from 15 to 2. It should not take 15 seconds for the server to respond anyway.

arendst, 2017-01-07:
Solved in version 3.1.13 using define USE_MQTT in user_config.h

arendst, 2017-01-08:
Added in version 3.1.13 just released. Disable MQTT using define USE_MQTT in user_config.h

closed 2017-01-08 by arendst