arendst/Sonoff-MQTT-OTA-Arduino #177: Home Assistant: Status of the switch stays ON after restart

ozeebrown, 2016-12-31:
Hi. I have setup a sonoff dual switch into my homeassistant successfully. Thank you.

But I have a small issue.

When both switches on sonoff are OFF (see image) and I restart homeassistant, the homeassistant interface for Switch No.2 turns and stays ON. This is a false state display in homeassistant.

[image]

Is this someone here had already picked up? Could the admin can give me an idea for a solution please.

Thank you. Kind Regards. Guy.

eifinger, 2016-12-31:
I would suggest you do a mosquitto_sub -t stat/# upon restart of your HA to See what it is really receiving. From there on we can narrow down on your issue.
How does your HA config for the two switches look like?

eifinger, 2016-12-31:
Sorry, Command (on Linux command line) should be "mosquitto_sub -v -t stat/#"

ozeebrown, 2016-12-31:
Thanks.

mosquitto_sub -v -t stat/# -u pi -P xxxx
does not show an update when homeassistant restarts.

What ever the status of the dual switch (before ha restart), homeassistant always change status to SWITCH 1 OFF and SWITCH 2 ON (after restart).

Apart from this condition, everything else on ha with dual sonoff switch works fine.

  - platform: mqtt
    name: "Dual light 1"
    state_topic: "stat/HAS01/POWER1"
    command_topic: "cmnd/HAS01/POWER1"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true
  - platform: mqtt
    name: "Dual light 2"
    state_topic: "stat/HAS01/POWER2"
    command_topic: "cmnd/HAS01/POWER2"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

arendst, 2016-12-31:
Independent from Home Assistant you should see retained messages with the ```mosquitto_sub -v -t stat/#``` command. If not, you have retained messages turned off on your sonoff dual.

Just execute ```powerretain on``` and toggle both switches. When mosquitto_sub is now executed again you must see two retained messages.

ozeebrown, 2016-12-31:
Thank you for the quick reply.

I have the sonoff (single replay) flashed using https://github.com/KmanOz/Sonoff-HomeAssistant working fine on the same homeassistant.

  - platform: mqtt
    name: "Patio Lights"
    state_topic: "home/sonoff/living_room/1/stat"
    command_topic: "home/sonoff/living_room/1"
    qos: 1
    payload_on: "on"
    payload_off: "off"
    retain: true

The single replay sonoff remembers the previous state and reflect manual button and ha switch together correctly.

This is what I get for the command looking at the dual switch sonoff ...
pi@raspberrypi:~ $ mosquitto_sub -v -t stat/# -u pi -P xxxx
stat/sonoff/POWER1 ON
stat/sonoff/POWER2 on
stat/HAS01/POWER1 OFF
stat/HAS01/POWER1/stat on
stat/HAS01/POWER1/ on
stat/HAS01/POWER2 ON
stat/HAS01/POWER2/RESULT ON
stat/HAS01/POWER2/ on
stat/HAS01/POWER2/stat on
stat/HAS01/POWER2 =  on
stat/HAS01/POWER2 = /RESULT OFF
stat/HAS01/POWER1 =  on
stat/HAS01/POWER1 = /stat ON
stat/HAS01/POWER1  on
stat/HAS01/POWER2  on

arendst, 2016-12-31:
And you see there is the stat/HAS01/POWER2 ON message as reflected by HA. This message is send from your broker NOT sonoff and is remembered by a previous retained message. I guess you have sonoff retain messages off.  As said before enable sonoff retain by execting command ```powerretain on``` and flip some switches. Then this message should update.

ozeebrown, 2016-12-31:
Thank you. the command powerretain on fixed that issue

17:07:05 MQTT: stat/HAS01/POWER1 = ON (retained)
17:07:08 MQTT: stat/HAS01/POWER2 = ON (retained)
17:07:47 MQTT: stat/HAS01/POWER1 = OFF (retained)
17:07:58 MQTT: stat/HAS01/POWER2 = OFF (retained)

What does the following in user_config.h? I have tried it, but it does work as the manual command.
#define MQTT_POWER_RETAIN      0

I have found a different issue now. As per the above image, I have grouped the switches. 

When I use the Group Switch ON, then the UI indicates all switches to be ON.

But only Switch 1 turns ON in sonoff

Then, Group switch OFF, UI indicates all switches OFF

sonoff Switch 1 is OFF, Switch 2 is ON

Toggle Group Button makes the sonoff Switch 1 ON with Switch 2 OFF / Switch 1 OFF with Switch ON

This happens when I use a group button trying to switch both switch ON/OFF at the same time.

Working with individual switches on HA UI works well.

Hope what I wrote make sense. Thank you for your help.

thomasf, 2016-12-31:
If you don't want to use retained messges, home assistant can be configured to request the sonoffs state on start up..

Something like this:

```yaml
  - alias: 'HA_Start'
    hide_entity: True
    trigger:
      platform: event
      event_type: homeassistant_start
    action:
      service: mqtt.publish
      data:
        payload: ""
        topic: "cmnd/sonoffs/power"
        qos: 0
        retain: 0
```

ozeebrown, 2017-01-01:
Thanks. "powerretain on" fixes the first issue.

But could not find a solution to the Switch 2 works inversely when using the Group Button.

To me, it looks like the Dual sonoff Switch can not accept Switch 1 and Switch 2 ON/OFF (group) command at once.

I wish there is a way to keep a delay between devices in Group Feature in Homeassistant.

I have ordered a 4 way sonoff Switch from itead.cc. Hope to investigate a little further soon.

Has anyone tried Dual Sonoff with HA using this firmware and using a group feature successfully?

Thank you. Regards, Guy.

arendst, 2017-01-01:
And another command that might help is seriallog 0.

ozeebrown, 2017-01-02:
powerretain on
seriallog 0

... is the solution.
What does seriallog do, in this situation?

Thank you very much for sharing.

arendst, 2017-01-02:
Glad it works. I've updated the wiki with a Sonoff Dual section explaining the need to disable serial logging.

closed 2017-01-02 by arendst