arendst/Sonoff-MQTT-OTA-Arduino #178: Support of "1 Channel Inching /Self-Locking WiFi Wireless Switch 5V 12V"

lionhe1966, 2016-12-31:
Do you plan to add support for this device ?
https://www.itead.cc/inching-self-locking-wifi-wireless-switch.html
I've read on itead forum that someone was able reflash it, even though there was a problem with the header pins.
http://support.iteadstudio.com/support/discussions/topics/11000009218

 I've just ordered one to control a gas boiler.
Thank you,

arendst, 2016-12-31:
See #159. Seems to work.

lionhe1966, 2016-12-31:
Thank you for your answer. I updated the wiki to mention also this device. I'll test as soon as I will receive it.

closed 2017-01-09 by arendst

lionhe1966, 2017-01-29:
I recently bought from Banggood this switch and I found that this firmware works but the RELAY_PIN is connected to GPIO13 instead of GPIO12.  Maybe there are different versions of this switch around.

putos11, 2017-08-28:
@lionhe1966 ,,
how did you get into Flash/programming mode??

lionhe1966, 2017-09-08:
@putos11
I basically followed the  wiki instructions: if i remember well,  you have to connect the ftdi programmer to the board,  press a button and then  connect the ftdi programmer to the pc.