arendst/Sonoff-MQTT-OTA-Arduino #179: Question: LWT sending "Offline" but newer "Online"?

fiskfan, 2016-12-31:
First. Thank you for all the work you have put into this project. - Is there a way to donate?

Question. The device sends the LWT topic with message "Offline" and it seems to work when the device goes offline. But i can't find a way for the device to send a "Online" message in the LWT topic when it goes online. - Is there a way to configure your code so that it does send an online message in the LWT topic?

arendst, 2016-12-31:
LWT stands for Last Will and Testament. A nice feature of MQTT where the MQTT server sends this message when the device has lost contact with it.

When a baby is born you do not expect a LWT but a nice birthday announcement. That's what sonoff does too. You'll receive at least two stat/sonoff/INFO messages when it comes to life.

That's how it is and should be.

timtimsson, 2016-12-31:
mqtt_publish(/LWT , "connected", true);
?

arendst, 2016-12-31:
Ok, just read over the MQTT info. I will implement it...

timtimsson, 2016-12-31:
... maybe as "one word":
/tele/ESP_F02/LWT Online
so you can better analyze the message in openhab?
(count online or offline :)

fiskfan, 2016-12-31:
Currently im using the "LWT" output to show the device state in node-red dashboard. I can use a function that overwrites the "LWT" Offline message when the device sends for example its uptime. But i think it would have been "nicer" if the LWT as "online" when the device comes online since it then removes the extra node-red funtction.

In my other project i use the "Connected"/"Disconnected" Last will messages. - But maybe i have used it incorrectly.

arendst, 2016-12-31:
Implemented as ```tele/sonoff/LWT Online``` (or as ```tele/sonoff/TELEMETRY {"LWT":"Online"}``` if JSON is enabled) in version 3.1.7 to be released this or next year ;-)

timtimsson, 2017-01-01:
Thanks,
.....but if we use mosquitto.conf >persistence true
then LWT/Offline not override tele/xxx/LWT Online 
You can check it with restart mosquitto and subscribe tele/#

please maybe use:
 (mqttClient.connect(MQTTClient, sysCfg.mqtt_user, sysCfg.mqtt_pwd, stopic,  **1, 1**, svalue)

arendst, 2017-01-03:
Updated and tested in version 3.1.8 just released