arendst/Sonoff-MQTT-OTA-Arduino #181: Need better documentation for Sonoff Pow messages

tavalin, 2016-12-31:
I've started to integrate my Sonoff Pow with openHAB. I've set up a rule that sends a message to the STATUS topic with payload "8" and I receive a response as follows:

`{"StatusPWR":{"Voltage":222, "Current":"1.067", "Power":234, "Today":"0.048", "Factor":"0.98"}}`

Can we update the wiki/documentation to describe the values above please.  Voltage and Power seem fairly obvious, but I'm unsure what Current and Today are and how they are calculated. (I thought "Today" was energy used so far in kwh, but when I reset my router it also seems to have reset to 0).

arendst, 2016-12-31:
Set messageformat to 0 and mqttunits to 1 and you'll see what the fields mean. Afterwards just set messageformat back to 1 again.

Today is the energy used starting at 00:00 until now. It's value is lost after a reboot as otherwise your flash would be gone within a few month while storing current values. That was what happened in your router reboot case. Stop sonoff reboots on wifi errors by setting wificonfig to 4

I use domoticz which stores today updates in it's database so my daily updates are not lost.

arendst, 2017-01-01:
Updated the wiki with a Sonoff Pow section which hopefully gets you on track.

tavalin, 2017-01-01:
I've just had a quick look through, that's VERY much appreciated :)

closed 2017-01-01 by tavalin