arendst/Sonoff-MQTT-OTA-Arduino #182: Remote configure Sonoff type?

smadds, 2017-01-01:
Would it be possible to configure the Sonoff board type and sensor after installing the software? It would be more convenient to have one binary image with the configurations stored in the unit's flash if there is space to compile the different options.

(I'm asking after flashing a Dual and a TH16 with updated software and realising the TH is not now recognising the sensor)

arendst, 2017-01-01:
That's what OTA is for.

smadds, 2017-01-02:
But isn't local flash used for keeping configuration so that the codebase
does not need editing each time there is an update? What's the point if we
still need to edit it to compile different binaries for the sonoff
variations?

Or am I missing something?

ecsfang, 2017-01-02:
I interpreted @smadds question to why we don't have a single binary, one to rule them all, and then do all configuration on each device?
That way I could OTA upgrade all devices (being Basic, TH, Pows or whatever) with one single binary.
And in the best of worlds, the binary itself should detect the hardware (number of relays, temp sensors etc), but that might be to set the hurdle to high ... ;)
Just my 2 cents, cheers!

arendst, 2017-01-02:
Asking a question is answering it.

Having all we want in one image is what we all want but:
- some things exclude another
- 1M flash for program storage means making choices about functionality and user experience
- 64k parameter storage lowers usable flash size
- OTA cuts useable flash size in half

Current image with TLS is already challenging and approaching max program size.

So as always there are catches and choices to make but having OTA anything is possible...

smadds, 2017-01-03:
I can understand that, especially with the TLS option. 

Would still be great to enable sensors without re-flashing though...

arendst, 2017-01-06:
Working on some kind of 'one size fits all'...

arendst, 2017-01-15:
Have a look in the experimental folder where I released a beta of the way I would like to go forward. Pls read the readme too.

smadds, 2017-01-16:
This is looking really good!
At the moment, I don't seem to be able to communicate with my MQTT server. I go into the web server settings (in between the long timeouts for MQTT) and set the new server, but it does not connect to my MQTT server. Has anyone else had this issue with the experimental version?

arendst, 2017-01-27:
NextGen Version 3.9.3 contains a lot of Domoticz updates like defining sensor indexes to supported sensors. See Domoticz Configuration webpage for more information.

closed 2017-01-27 by arendst