arendst/Sonoff-MQTT-OTA-Arduino #185: tele/#/Uptime

jaysuk, 2017-01-02:
This always seems to return 0 on a sonoff device, regardless of how long it has been "up".
Have tried it with both power to the relay and without.

arendst, 2017-01-02:
I'm swamped with uptimes:
```
tele/strip1/UPTIME = 126
tele/pow1/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/dual/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":50}
tele/wkaku12/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/wkaku10/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/sonoff1/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/sonoff3/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/wkaku11/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/bridge868/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":118}
tele/wkaku8/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/th1/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/bridge434/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":115}
tele/wkaku9/TELEMETRY = {"Time":"2017-01-02T16:02:30", "Uptime":125}
tele/ring1/UPTIME = 126
```

jaysuk, 2017-01-02:
I assumed uptime would be in minutes.
Is it in hours?

arendst, 2017-01-02:
Yes it is in hours.

closed 2017-01-02 by jaysuk