arendst/Sonoff-MQTT-OTA-Arduino #186: OTA upload Arduino 1.6.13 [ERNO 2] Problem

timtimsson, 2017-01-02:
Hello,
i've a problem with compiling ... i think it's a path problem - how can help me to solve this?
(Windows10, python 2.7 installed, Arduino 1.6.13, espupload.py->tools, boards.txt and platform.txt->2.3.0) 

> python.exe: can't open file 'C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0<b>/</b>tools<b>/</b>espupload.py': [Errno 2] No such file or directory
> Beim Hochladen des Sketches ist ein Fehler aufgetreten

timtimsson, 2017-01-02:
ahhh my faulty: espupload.py->\esp8266\2.3.0\tools
but now:
16:34:33 [DEBUG]: Options: {'debug': True, 'host_port': 80, '<b>image</b>': 'C:\\Users\\Tom\\AppData\\Local\\Temp\\arduino_build_862925/sonoff.ino.bin', 'host_ip': '10.1.1.200'}

  <h1>417 - Expectation Failed</h1>

arendst, 2017-01-02:
Google is your friend regarding 417 - Expectation Failed.

I suppose you have set-up a webserver and prepared the api directory as advised in the wiki where the receiving php script is stored.

timtimsson, 2017-01-02:
.. yes, i prepared the folder api with the original upload-arduino.php 
and a empty r/w folder: /api/arduino
if i test the server: http://10.1.1.200/api/upload-arduino.php
-> Sorry, there was an error uploading your file to OTA server 10.1.1.200.
at espupload.py i changed only the ip:
LOCALADDR = "10.1.1.200"

What could be wrong?

arendst, 2017-01-02:
The error message should name the file you want to upload. For some reason the filename is not received. Have a look at the small file in api folder.

timtimsson, 2017-01-02:
... i don't have a file in the api folder besides upload-arduino.php ?

arendst, 2017-01-03:
That's as it should be. If you test the server with http://10.1.1.200/api/upload-arduino.php the result is as expected too as you did not provide the firmware image to the program. At least you now know you have a functioning PHP environment.

Next step is testing the functionality of the ```espupload.py``` program by opening a command prompt at ```C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\tools\``` and execute command ```espupload.py -f espupload.py```. This should upload the file to your webservers' ```api/arduino``` folder and respond with the message ```The file espupload.py has been uploaded to OTA server 10.1.1.200.```

timtimsson, 2017-01-03:
Thanks for your patience.
unfortunately the same problem:
`C:\Users\Tom\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\tools>python espupload.py -f espupload.py
11:03:11 [DEBUG]: Options: {'debug': True, 'host_port': 80, 'image': 'espupload.py', 'host_ip': '10.1.1.200'}`
`417 - Expectation Failed`

I use  a RaspberryPi as server and i installed lighttpd, PHP5.6.27 and python 2.7...
my pi upload works with python and php in the api folder
[test.zip](https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/files/682056/test.zip)

but if i check espupload.py -f espupload.py at the pi - the same error 417!?
Should i install apache at pi?

arendst, 2017-01-03:
Googling found this:
```
417 errors in the HTTP cycle

Any client (e.g. your Web browser) goes through the following cycle when it communicates with the Web server:
1.Obtain an IP address from the IP name of the site (the site URL without the leading 'http://'). This lookup (conversion of IP name to IP address) is provided by domain name servers (DNSs).
2.Open an IP socket connection to that IP address.
3.Write an HTTP data stream through that socket.
4.Receive an HTTP data stream back from the Web server in response. This data stream contains status codes whose values are determined by the HTTP protocol. Parse this data stream for status codes and other useful information.

This error occurs in the final step above when the client receives an HTTP status code that it recognises as '417'.
```
I'm afraid your webserver somehow fails to respond correctly. Maybe you can find more info specific to lighttpd and 417.

arendst, 2017-01-03:
I've just uploaded a patched version of the espupload.py in the Experimental repository. Give it a try and let me know the results.

timtimsson, 2017-01-03:
:) nearly good...
because the posted file is the right with wrong name (/api/Arduino/`Array`)

arendst, 2017-01-03:
That's because your arduino folder in api on your webserver starts with an uppercase A?

timtimsson, 2017-01-03:
oh sorry no - copy fail
`/api/arduino/Array`

both Windows and raspberry have the same defect...

arendst, 2017-01-03:
What file are you trying to upload?

On your webserver in file upload-arduino.php remove the two slashes before line 4 (var_dump) and try to upload a file again. What response do you get. Mine is:
```
array(1) {
  ["file"]=>
  array(5) {
    ["name"]=>
    string(12) "espupload.py"
    ["type"]=>
    string(24) "application/octet-stream"
    ["tmp_name"]=>
    string(14) "/tmp/phpss7Hmi"
    ["error"]=>
    int(0)
    ["size"]=>
    int(2683)
  }
}
The file espupload.py has been uploaded to OTA server domus1.
```

arendst, 2017-01-03:
Verify your php.ini file for allowing HTTP file uploads:
```
;;;;;;;;;;;;;;;;
; File Uploads ;
;;;;;;;;;;;;;;;;

; Whether to allow HTTP file uploads.
; http://php.net/file-uploads
file_uploads = On

; Temporary directory for HTTP uploaded files (will use system default if not
; specified).
; http://php.net/upload-tmp-dir
;upload_tmp_dir =

; Maximum allowed size for uploaded files.
; http://php.net/upload-max-filesize
upload_max_filesize = 2M

; Maximum number of files that can be uploaded via a single request
max_file_uploads = 20
```

timtimsson, 2017-01-03:
array(1) {
  ["file"]=>
  array(5) {
    ["name"]=>
    string(12) "espupload.py"
    ["type"]=>
    string(24) "application/octet-stream"
    ["tmp_name"]=>
    string(14) "/tmp/phpkqv9Ty"
    ["error"]=>
    int(0)
    ["size"]=>
    int(2689)
  }
}
The file Array has been uploaded to OTA server homeserverpi.

arendst, 2017-01-03:
That looks very good. I'm puzzled at the moment. Is the file present in your api/arduino folder?

timtimsson, 2017-01-03:
yes, there is the right file with wrong name - "only"

php.ini is the same like you....

arendst, 2017-01-03:
Try to change the basename line as below:
```
//$image = basename($_FILES["file"]["name"]);
$image = $_FILES["file"]["name"];
```
and see what happens...

timtimsson, 2017-01-03:
[image]
perfect - thanks a lot!

arendst, 2017-01-03:
I have no idea why basename doesn't work for you. Anyway we have a workaround.

timtimsson, 2017-01-03:
Hi arendst,
i flashed my Raspberry new...
`$image = basename($_FILES["file"]["name"]);` works now !!!
No idea what the problem was...
P.S. `$image = $_FILES["file"]["name"];` also works on the "new"pi image...

closed 2017-01-03 by timtimsson