arendst/Sonoff-MQTT-OTA-Arduino #187: How do I change the relay pin

kwittwer, 2017-01-02:
Hi, 
I bought one of these : https://github.com/JanGoe/esp8266-wifi-relay 
I try to change the relay pin to GPIO2 to be able to switch relay 1 of the board but it didn't worked. Can you please provide a hint how to do this. I used the "sonoff" board type an changed REL_PIN from 12 to 2. 
thank you

arendst, 2017-01-02:
Try the other relay on gpio14

kwittwer, 2017-01-02:
I have bought the version with just 1 relay on it, so I can just try this one. But I will try to change relay2 to pin 2. Is the way I tryed it correct?

arendst, 2017-01-02:
Yes I guess it could work. Just compare both schematics.

It might be that my software, by lack of a relay at gpio12 has selected the dual version. Try to execute command model 1 and see what happens.

kwittwer, 2017-01-03:
Good morning, with the command model 1 it works great. Is ther a way to set this by using a compilerswitch or soemthing?

arendst, 2017-01-03:
No, it's part of the autodetection circuit I use to detect if the hardware is a Sonoff Basic, A dual or a 4CH.

People are already complaining they have to compile to much for every hardware change...

If you had a resistor to Gnd on the relay pin 2 it would have worked perfectly well.

I think having the model command should make most people happy.

closed 2017-01-03 by arendst

kwittwer, 2017-01-03:
ok adding a risistor is ok for me, I will try it.  Is the model command saved to flash?

arendst, 2017-01-03:
Yep. Once detected it is saved to flash and as long as there is no ```reset 1``` executed or user_config.h CFG_HOLDER change being made the auto-detect circuit will not be run again. 

So I would say forget the resistor and just remember to execute ```model 1``` after major changes.