arendst/Sonoff-MQTT-OTA-Arduino #188: Delay timer ?

F6GNJ, 2017-01-03:
Hello Theo, all,

First bravo for this very good and usefull software! I use an RF sonoff module for external lighting and it would be interesting to have a timer of the lighting of the projector. This avoids forgetting to turn off the light ... with this automatic shut-off. I use the RF version and it works very well. An item in the WEB server will allow to program the duration of operation in seconds 0 = normal operation without time delay and 65535 = +/- 45 days ... This could also constitute a safety on a lighting or heating system?
Regards,
Arnaud.

arendst, 2017-01-03:
Will investigate

arendst, 2017-01-03:
It's already present for relay 1.

Just use command PulseTime to set a time in 0.1 Sec increments to allow for the relay to be On. See the commands section in the wiki.

closed 2017-01-04 by arendst

F6GNJ, 2017-01-04:
Hi Theo,

 Great job, thanks for time and enhancement !

Regards

Arnaud.