arendst/Sonoff-MQTT-OTA-Arduino #191: Time Zone not updating

imtiajmeah, 2017-01-04:
Hi,

I'm sure im being stupid so forgive the noob question. I have updated the user_config.h file with the below changes to reflect UK(London) time, but after a restart it still seems to be 1 hour ahead. Any idea what im missing or did incorrectly.

// -- Time - Up to three NTP servers in your region
#define NTP_SERVER1            "pool.ntp.org"
#define NTP_SERVER2            "nl.pool.ntp.org"
#define NTP_SERVER3            "0.nl.pool.ntp.org"

// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST               Last, Sun, Mar, 2, **+60**  // Last sunday in march at 02:00 +60 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD               Last, Sun, Oct, 3, **+0**   // Last sunday in october 02:00 +0 minutes

// -- Application ----------------------------
#define APP_TIMEZONE           **99**           // [Timezone] +0 hour (London) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)

arendst, 2017-01-04:
Try command ```timezone 99``` as you didn't tell user_config.h to use the new settings as stated in the very first lines of the file:
```
/*********************************************************************************************\
 * User specific configuration parameters
 * 
 * ATTENTION: Changes to most PARAMETER defines will only override flash settings if you change
 *            define CFG_HOLDER. 
 *            Most parameters can be changed online using commands via MQTT, WebConsole or serial**
 *
 * Corresponding MQTT/Serial/Console commands in [brackets]
\*********************************************************************************************/
```

imtiajmeah, 2017-01-04:
That worked ;)

Thanks for the amazing work you are doing... 

Just got my first Sonoff today and playing around with your amazing firmware.

closed 2017-01-04 by imtiajmeah