arendst/Sonoff-MQTT-OTA-Arduino #192: Is WEMO_EMULATION implemented ?

kiwipaulrob, 2017-01-04:
I have enabled it in user_config.h but Alex is not finding it as new device. Also is there a place to define the device name for Wemo purposes?  

Thanks in advance.

arendst, 2017-01-04:
Search for alexa in this site. MQTT_CLIENT_ID is used as alexa friendly name.

gadjet, 2017-01-04:
I got my light working with Alexa (Slampher) but one thing I've noticed is that when you use a space in the MQTT_CLIENT_ID, I used "Light 1" as the name but Alexa will not respond to "light 1" but when I just say "Light" it operates it OK.

I will try giving it a name without a space then a number or two words with a space like "Light one"

samsixtysix, 2017-01-04:
I couldn't get my UK based Alexa to find any device either with option enabled and I tried a couple of different MQTT Client IDs.

Alexa did find devices when running them through HA-Bridge but not with the built in Wemo emulation.

kiwipaulrob, 2017-01-05:
I am in NZ and https://bitbucket.org/xoseperez/fauxmoesp sketches work. Not sure where to start debugging my problem with this code. FYI I am not running this on a Sonoff (waiting for one) but don't think that this is the problem.

samsixtysix, 2017-01-05:
I got mine working - maybe obvious for most but for anyone else struggling

I needed to do the following:
set MQTT_CLIENT_ID to the device name I wanted Alexa to respond to
uncomment USE_WEMO_EMULATION
change the value of CFG_HOLDER (I changed it to '1')
and have the sonoff connect to an MQTT broker

Without the MQTT broker running Alexa could not find the sonoff but once found she could continue to turn it on or off with or without the MQTT broker running.

psyc0, 2017-01-06:
Works for me in SE, as mentioned by @samsixtysix the broker needs to be online for this to work (noticed since I readdressed the broker whilst flashing this repo.)

psyc0, 2017-01-07:
Just for information, just reflashed 4 standard sonoff´s , for each time I added a new to the network I lost(offline) all previous and Alexa would not find the new either.. the trick for me was to powercycle the echo dot when I was done. problem now is that she finds 3/4, and if I search again she looses all the others and I need to reboot the echo..

closed 2017-01-08 by arendst