arendst/Sonoff-MQTT-OTA-Arduino #196: Timezone problem

macadamer, 2017-01-05:
I believe I have correctly set up user_config.h for UK timezone (we're GMT at the moment), however times returned are 1 hour ahead. Where have I gone wrong please?

Reported by status mqt:

14:46:13 MQTT: stat/sonoff/RESULT = TIM: UTC Thu Jan 05 13:46:13 2017, Local Thu Jan 05 14:46:13 2017, StartDST Sun Mar 26 01:00:00 2017, EndDST Sun Oct 29 02:00:00 2017

And the section from user_config.h:

// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST               Last, Sun, Mar, 1, +60  // Last sunday in march at 01:00 +60 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD               Last, Sun, Oct, 2, -60   // Last sunday in october 02:00 -60 minutes

// -- Application ----------------------------
#define APP_TIMEZONE           0            // [Timezone] +0 hour (London) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)

allthepies, 2017-01-06:
I also had this same issue but as the above comment you have to change the CFG_HOLDER #define value to get the NTP pararms changed.

macadamer, 2017-01-06:
I see my error confusing absolute and relative time settings, but I can't decipher CFG_HOLDER to apply the needed override for auto DST/STD changes!

allthepies, 2017-01-06:
Just change the CFG_HOLDER value, there's no deciphering to be done :)  Add one to it for example.

macadamer, 2017-01-06:
I had just come to the same conclusion, after searching through the cleared issues! It's working now.

closed 2017-01-06 by macadamer

oywino, 2019-12-04:
Guys - exactly how did you solve this?
My problem is that the timezone in Mosquitto is UTC and I want it to be ECT (Europe/Brussels).
I'm running Mosquitto v1.6.8 and it seems to ignore the TZ environment variable.
Any help is apprechiated.