arendst/Sonoff-MQTT-OTA-Arduino #197: GPIO14 MQTT Status only no relay control

stoehrmark, 2017-01-05:
I know you have answered this before but I cannot figure it out. 
I want to use GPIO14 with a switch. I want to know if the switch is open or closed (1 … 0). I do not want the switch to control the relay on GPIO12 just show its status to MQTT. Below is the circuit I am using. I have a DHT22 connected to GPIO3 (this is working fine). I have commented out the part in the script that won’t allow 2 thing on GPIO14 (see below). When i use the code as is the relay turns on and off. Not what i want.

How do I do the following that you posted --> 
`Define a ButtonTopic and you'll receive a cmnd/<ButtonTopic>/POWER message when USE_WALL_SWITCH is enabled.`

Thanks for all the work on this project.

/*-------------------------------------------------------------------------------------------*\
 * DHT11, DHT21, DHT22, AM2301, AM2302 and AM2321
\*-------------------------------------------------------------------------------------------*/
  #define DHT_PIN              3           // GPIO 14 = AM2301 (Sonoff_TH10A(16A), Sonoff SV)
  #define DHT_TYPE             DHT22       // DHT module type (DHT11, DHT21, DHT22, AM2301, AM2302 or AM2321)
  // *** Option 1 - No external library needed - Select either Option 1 OR Option 2
 #define SEND_TELEMETRY_DHT                // Enable sending temperature and humidity telemetry
  // *** Option 2 - Use Adafruit DHT library - Select either Option 1 OR Option 2
//  #define SEND_TELEMETRY_DHT2               // Enable sending temperature and humidity telemetry
/*-------------------------------------------------------------------------------------------*\
/*-------------------------------------------------------------------------------------------*\
 * Wall switch and HC-SR501
\*-------------------------------------------------------------------------------------------*/
  #define SWITCH_PIN           14           // GPIO 14 = Standard wall switch to Gnd (Sonoff_TH10A(16A), Sonoff SV)
  #define SWITCH_MODE          FOLLOW       // [SwitchMode] TOGGLE, FOLLOW, FOLLOW_INV, PUSHBUTTON or PUSHBUTTON_INV (the wall switch state)
  #define USE_WALL_SWITCH                   // Enable the use of a standard wall switch to control the relay
/*-------------------------------------------------------------------------------------------*\

I have commented out the following because I use GPIO3 for the DHT 
/*
#if ((defined(ST_DSB) && defined(ST_DHT)) || (defined(ST_DSB) && defined(USE_WALL_SWITCH)) || (defined(ST_DHT) && defined(USE_WALL_SWITCH)))
#if ((DSB_PIN == DHT_PIN) || (DSB_PIN == SWITCH_PIN) || (DHT_PIN == SWITCH_PIN))
  #error "Select either SEND_TELEMETRY_DS18B20 or SEND_TELEMETRY_DHT or USE_WALL_SWITCH or use different GPIOs"
#endif
#endif
*/

[image]

[image]

alphalove, 2017-01-05:
Hello,
Believe this is possible, have a look at https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/141

Cheers,

alex

arendst, 2017-01-06:
Execute command ```ButtonTopic test1```. 

When you now flip the switch a message should be send by sonoff like ```cmnd/test1/POWER ON```.

stoehrmark, 2017-01-06:
OK so in console on the web interface i entered ButtonTopic garagedoor . I now get a response of change from the Sonoff 

Switch ON  --> 09:04:29 MQTT: cmnd/garagedoor/POWER = ON
Switch OFF --> 09:04:30 MQTT: cmnd/garagedoor/POWER = ON

2 Questions. 

1. The Sonoff on board button no longer works as before. It is now assigned to the ButtonTopic (probably this is what you expect). Can this be a separate thing? The on board button GPIO0 working as normal and a switch attached to GPIO14 assigned to the ButtonTopic independent of each other.

2. The response from ButtonTopic  is only ON. Is there a way to get ON and OFF status
e.g.
Switch ON  --> 09:04:29 MQTT: cmnd/garagedoor/POWER = ON
Switch OFF --> 09:04:30 MQTT: cmnd/garagedoor/POWER = **_OFF_**

arendst, 2017-01-06:
Your findings are correct. I'll investigate in the future.

stoehrmark, 2017-01-06:
Thanks for your work on this. It seems that a few people have been asking for a similar feature. Can't wait until you have time to add it in. All the best and thank you again.

arendst, 2017-01-07:
Solved in version 3.1.14 just released where it would send a TOGGLE. See #206 for explanation.

arendst, 2017-01-08:
See #206.

closed 2017-01-08 by arendst