arendst/Sonoff-MQTT-OTA-Arduino #198: Generic Sonoff Profile - Feature request?

jmtosses, 2017-01-05:
I'm wondering if a generic Sonoff profile could be a good idea. Please excuse if this has been discussed before.

The idea would be to be ready for anything Sonoff throws at us before the new profile is committed to the project. Or ... in my specific case I would like to use this firmware for all my ESP based IoT devices and a generic profile could be a good approach to achieve this.

Thank you for considering.
Cheers! JM

arendst, 2017-01-06:
Generic profile ?!? How do you envision this? Sonoffs come in all kinds of configuration using ESP8266, ESP8285, PSB etc.

Please elaborate a bit more about your request.

jmtosses, 2017-01-08:
Hi,

yes, that was a bit of a shot from the hip. So let me elaborate a little bit on what I would love to see in the Sonoff-MQTT-OTA-Arduino file ...

I'm using - like so many here - various ESP and Arduino modules to perform different IoT and home automation tasks - all of those are running custom software either based on my own code or for example frameworks like mySensor. Just very recently I added a few Sonoff devices to the mix and of course I love them as they feature out of the box what took me many nights to build myself - for a very comparable price.

After watching your firmware being featured on SuperHouse on Youtube I started fimiliarising myself with it and I really love what it does and how it does it. In short it could be the go-to firmware for many if not most of my ESP based projects.

To make this less hacky I would suggest to add a configuration just for the ESP8266 featuring all it's GPIOs to configure for usage with the libraries you already link to like onewire, DS18B20, DHTnn and the like plus add some button options as actuators/wall switches.

So instead of choosing "Sonoff" in the configuration you could choose "generic" or "ESP8266" for home brew hardware utilising your fine software.

What do you think?

arendst, 2017-01-08:
I think I see what you mean. I've already started with something more GPIO based than the current function based user interface. Problem is that it soon grows to large and over 450k which would be the limit for using OTA on 1M flashes as the sonoff are using.

On a side track I'm working on it but for now the current branch will be maintained and upgraded.

Hogesyx, 2017-01-13:
I think you might be looking for [ESPEasy](http://www.letscontrolit.com/wiki/index.php/ESPEasy).

jmtosses, 2017-01-13:
Thank you Hogesyx,

I've heard about ESPEasy but didn't test it yet. Will test it.

Best regards,
JM

Hogesyx, 2017-01-13:
That project is pretty great for playing with various sensors, however if you want over the air firmware upgrade you need to replace the 8Mb flash with something larger on the sonoff. My initial HA was actually started with that.

This Sonoff-MQTT-Arduino project on the other hand is much more useful for bulk deployment of multiple similar sonoff. As I like to have hard coded config in binaries so anytime a device is dead I just need to reflash back the same firmware and setup basic naming and the system will be back on. But that is just my work flow so YMMV.

There is no harm using a combination of both for a complete home automation setup.

jmtosses, 2017-01-14:
Hogesyx, I gave it a spin and I like what I see. It will be a combination of both as five more Sonoff have arrived today. Cheers!

arendst, 2017-01-15:
Have a look in the experimental folder where I released a beta of the way I would like to go forward. Pls read the readme too.

arendst, 2017-01-27:
NextGen Version 3.9.3 contains a lot of Domoticz updates like defining sensor indexes to supported sensors. See Domoticz Configuration webpage for more information.

closed 2017-01-27 by arendst