arendst/Sonoff-MQTT-OTA-Arduino #199: Tips for Touch?

nigelb135, 2017-01-05:
I received 5 Sonoff Touch units yesterday. Flashing with v3.1.11 reports success using Platformio -- just attached the control PCB with pin header added and shorting GPIO 0 to ground.

I've been successfully flashing and OTA upgrading other Sonoff units since v2 but...  the serial monitor displays nothing after the flashing and no wifi activity is present.

Anyone else had success? Any known 'tricks' to get the Touch to work?

Help would be appreciated.

Nigel

Theo, your pace of development is awesome! I just finished upgrading 10 units to v3.1.9 from v3.1.5 only to discover v3.1.11 had been committed even as I worked! Thanks for the great work.

arendst, 2017-01-06:
Did you change to ESP8285 in the IDE when you compiled your code. I guess platformio failed to do so. See wiki section Compile and Upload regarding Sonoff Touch.

nigelb135, 2017-01-06:
Thanks for that information. I honestly thought I had searched everywhere for the answer that was there all the time.

Got the first one going now.

closed 2017-01-06 by nigelb135