arendst/Sonoff-MQTT-OTA-Arduino #200: Using Sonoff by http-commands only

JoeWiseman, 2017-01-05:
Hi,

first let me applogize for using your great work for what it's not intend to be ... I'm using a sonoff S20 with your firmware on a homatic system (without any MQTT or Domoticz). So I just the the http-commands to switch it on or off.

I know that this is not intend to be, but is it possible to switch off all network activities (MQTT, Domoticz) and use the the switch just by http-commands as described in the closed topic 89?
Maybe there is a command for this which I didn't see? Or maybe I could set a wrong server adress and set the time interval of connecting the server to a very high value??

Another question: is it possible just to get the content of the return-message of an command instead of getting the full http-code when using wget? I just need to get the status of the switch, but when using the command >> wget -q -O /dev/stdout 'http://192.168.1.232/cm?cmnd=power%20on' << I get the full page-script instead of the connamd-answer...

Hope you can help me out though it's not your tastk to do!

Thanks a lot in advance
Joe

allthepies, 2017-01-06:
I would second this request but from a different angle, I want to control my sonoff via Alexa (and this works) and have the MQTT interface optional.   Currently I have to point the sonoff at an MQTT broker which I have no need for.

arendst, 2017-01-07:
Added in version 3.1.13 just released. Disable MQTT using define USE_MQTT in user_config.h

allthepies, 2017-01-07:
Awesome, thanks.

closed 2017-01-07 by arendst

JoeWiseman, 2017-01-08:
Thanks a lot for your work!!! 

Though I acually have just one Sonoff S20 in use and I need this one very badly I don't want to run any experiences with that one, so I have a few questions:
-1- I never used the OS-update-option from the web interface... Do I just click it and the latest version will be installed?
-2- Is the new option USE_MQTT to be set in the config.h-file for compiling the new version or is it a new command for toggeling the MQTT-functionality on/off when using the switch?
-3- You wrote "Add command response to web command". Though I haven't use the update-function yet ... how is the command-resonse behavoiur?

Would be great to get some feedback.

Regards

arendst, 2017-01-08:
1. Give it a try! Just have your binary file compiled and know where to find it.
2. As long as define USE_MQTT is enabled MQTT software is compiled in and functional. If you disable it and compile MQTT is removed from the code and MQTT won't be possible without recompiling
3. Again, give it a try. It's functional for HTTP request like ```http://sonoff/cm?cmnd=power on```. The console functionality hasn't changed.

JoeWiseman, 2017-01-09:
Awesome!!!! If any company would give such a great support I will be the first to patronise ;-)
Thanks again a lot for the very fast reply and the great portion of patience with my rookie questions!