arendst/Sonoff-MQTT-OTA-Arduino #201: Failed to open COMn after File upload buffer miscompare" when OTA a modified firmware

badabing2005, 2017-01-05:
Hi,
First and foremost, awesome project, it's amazing how thorough, complete and easy to use it is, not to mention all the powerful configuration options.

My Sonoff is the basic Sonoff
Initially I made the mistake of configuring the Flash size as 512K (64K SPIFFS).
It flashed and worked fine with no issues.
As I was making some configuration changes, (and upgrading from 3.1.8 to 3.1.11)
I realized that I had made a mistake in the flash size selection.
Corrected it and tried to flash OTA

I ran to the same issue as described [here](https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/157): 

Reading the details, I figured ok, no problem, I'll just flash it using the serial and I'll be ok.
However after the failed OTA update, the device appears to be bricked.
During the serial upload, I get 

> error: Failed to open COM3
> error: espcomm_open failed
> error: espcomm_upload_mem failed
> error: espcomm_upload_mem failed

I know it's not the serial cable nor my method of serial flashing, as I successfully updated two other sonoffs from 3.1.8 to 3.1.11 with the corrected Flash size using only the serial flash method.

I'm assuming that either the failed OTA update (from the web server) caused the brick, or the device decided to go bonkers on me when I plugged in the Serial cable (PS: the device was operational prior to this)

Is there anything that can be done to revive this one unit?
How can I fix the com?
Or anything that helps me debug this?

Oh one other note, as soon as I plug the serial cable and the device gets powered (though LED does not come on), my USB Logitech mouse starts going berserk, it's all over the place, moving randomly and clicking randomly.
No issues, when serial cable is removed, or the cable is connected to a non-problematic Sonoff.
I'm assuming that this bricked device is causing some kind of RF interference (I suppose through 8266, as it does not have the RF module) 

Thanks in advance.

don-willingham, 2017-01-10:
The "Failed to open COM3" sounds like a serial port problem. Maybe another process is still using it. I would maybe reboot with only your mouse. Assuming it's a USB serial cable, watch your control panel as you plug it back in. Ensure it's enumerating it as COM3 again. I have two Sonoff dual(s) and I had to repeat my attempts at flashing.  I'm new to esp8266's and microcontrollers in general; but some flash attempts failed to complete, or others failed to start with errors like those.

smadds, 2017-01-10:
And on a Linux PC it can be worse - when I power up the Sonoff from my USB
to serial converter it resets the converter and often Linux assigns a new
serial port. There is a retired sequence for success:-

1) Power Sonoff with button pressed
2) Check serial port in Arduino IDE
3) Re-open serial monitor if port changed
4) Flash the code

badabing2005, 2017-01-12:
Thanks don-willingham and smadds for suggestions.

I'm beginning to get convinced that I have a hardware issue.
As mentioned I have 3 sonoffs, and the other two work fine, I unplug the bad one, plug the good, and it just works, no need for reboot or port check.

Regardless, I double checked, the port is ok, but somehow the communication with sonoff is not working.
Any other way to program this? JTAG perhaps?
Anyone has the schematics to follow where the TX, RX pins go, is there a serial chip on this that is possibly toasted?

Of course I don't care if this doesn't get revived, but it wouldn't be a hobby if I don't stab at fixing it.

Thanks

don-willingham, 2017-01-13:
Sounds like you have a dead one. There are schematics https://www.itead.cc/wiki/images/f/f8/Sonoff-Schmatic.pdf  I believe your programmer supplies the 3.3V and serial interface. It only needs to talk to the esp8266, so I am not sure if there is anything else on the board you can troubleshoot... Then again I'm better with software than hardware.

badabing2005, 2017-01-15:
Thank you Don,

I'll take a look at it and report back if I get anywhere with it.

closed 2017-02-04 by arendst