arendst/Sonoff-MQTT-OTA-Arduino #202: Socker error

aferraz, 2017-01-06:
Hi,
 after upgrade to  the last firmware my sonoff can't connect to mosquitto.
mosquitto side I have this error:
Jan  6 02:07:31 node mosquitto[11001]: New connection from 192.168.1.70 on port 1883.
Jan  6 02:07:31 node mosquitto[11001]: Socket error on client <unknown>, disconnecting

sonoff side this:
Jan  6 02:06:55 sonoff1-4534 ESP-MQTT: Attempting connection...
Jan  6 02:07:06 sonoff1-4534 ESP-MQTT: Attempting connection...
Jan  6 02:07:17 sonoff1-4534 ESP-MQTT: Attempting connection...

Any help ?

Thanks 
 Alessandro

arendst, 2017-01-06:
Try disabling mDNS dicovery in user_config.h and recompile and upload.
```
// -- mDNS -----------------------------------
//#define USE_DISCOVERY                       // Enable mDNS for the following services (+8k code, +0.3k mem)
  #define WEBSERVER_ADVERTISE               // Provide access to webserver by name <Hostname>.local/
  #define MQTT_HOST_DISCOVERY               // Find MQTT host server (overrides MQTT_HOST if found)
```

aferraz, 2017-01-06:
Hi arendst,
 Unfortunately I have the same error.

Regards
 Alessandro

arendst, 2017-01-06:
From the PubSubClient issue list:
```
OK,
 I found out that the error code 5 was from cloudmqtt " not authorised".

I had setup a new user "ESP8266Client" in cloudmqtt with a blank password, I'm guessing this isn't the way to do it.

Anyone successfully connected to Cloumqtt ? any pointers
```
Looks like you have MQTTUser or MQTTPasword error.

aferraz, 2017-01-06:
I use the same user and password in other client and it work.

arendst, 2017-01-06:
From the PubSubClient code:
```
// Possible values for client.state()
#define MQTT_CONNECTION_TIMEOUT     -4
#define MQTT_CONNECTION_LOST        -3
#define MQTT_CONNECT_FAILED         -2
#define MQTT_DISCONNECTED           -1
#define MQTT_CONNECTED               0
#define MQTT_CONNECT_BAD_PROTOCOL    1
#define MQTT_CONNECT_BAD_CLIENT_ID   2
#define MQTT_CONNECT_UNAVAILABLE     3
#define MQTT_CONNECT_BAD_CREDENTIALS 4
#define MQTT_CONNECT_UNAUTHORIZED    5
```

aferraz, 2017-01-06:
I know arendst, but:
11:08:14 RSLT: stat/sonoff1/RESULT = {"MqttHost","192.168.1.22"}
11:08:19 RSLT: stat/sonoff1/RESULT = ["MqttUser":"client"}
11:08:23 MQTT: Attempting connection...
11:08:28 RSLT: stat/sonoff1/RESULT = ["MqttUser":"client"}
11:08:36 MQTT: Attempting connection...
11:08:40 RSLT: stat/sonoff1/RESULT = {"MqttPassword":"EDITED"}
11:08:48 MQTT: Attempting connection...

aferraz, 2017-01-06:
Hi,
 solved after mosquitto reinstall

Thanks

closed 2017-01-06 by aferraz