arendst/Sonoff-MQTT-OTA-Arduino #205: Idea:Discussion Forum

tam481, 2017-01-06:
Apologies for raising this as an issue. It's more an idea.

Does anyone know a forum where we can discuss all @arendst and Sonoff related projects and ideas. I am particularly interested in how others are using various Sonoff devices around their homes etc.

rlust, 2017-01-08:
Good idea! How about a google group?

arendst, 2017-01-09:
You may start it and I might join in.

jaysuk, 2017-01-09:
Would you be happy with a google groups forum being set up?

tam481, 2017-01-09:
Shall I set it up? Happy to do so.

arendst, 2017-01-09:
tam481, be my guest

jaysuk, 2017-01-09:
Go for it. I can help moderate

tam481, 2017-01-09:
https://groups.google.com/d/forum/sonoffusers

tam481, 2017-01-09:
@jaysuk and @arendst please join in and I'll make you moderators :-)

arendst, 2017-01-13:
Links added to both Readme and wiki.

closed 2017-01-13 by arendst