arendst/Sonoff-MQTT-OTA-Arduino #206: Button not sending status to MQTT

osvaldoasn, 2017-01-06:
Hi, i need some help!
My sonoff + mqtt + homebridge is working great but when i push sonoff button the status does not change in the App (Home). When i activate it via app it works great!

HairBear22, 2017-01-06:
That's interesting! I have the exact same problem. What plugin for homebridge are you using?

osvaldoasn, 2017-01-06:
homebridge-mqttswitch

psyc0, 2017-01-06:
I think there is a bug in the MQTT_STATUS_OFF, while toggeling the relay the MQTT message sent every time was ON even if the relay was of. I wont open another ticket since I hope the problem will be solved here.
@osvaldoasn @HairBear22 have you verified with `mosquitto_sub -h #hostname -p 1883 -v -t '#'` that you actually see the correct messages on the broker?
when encountering this I had configured:

<pre>
MQTT_BUTTON_RETAIN = 1
MQTT_POWER_RETAIN = 1
MQTT_STATUS_ON = "ON"
MQTT_STATUS_OFF = "ON"
</pre>

psyc0, 2017-01-06:
Actually not 100% on what `MQTT_STATUS_OFF` does, just tried flashing one now and i still get the off status
MQTT_STATUS_OFF        "OFF"  
> Status result when turned off (needs to be a string like "0" or "Off")
<pre>
23:17:40 MQTT: sonoff-out/sonoff4/RESULT = {"POWER":"ON"}
23:17:40 MQTT: sonoff-out/sonoff4/POWER = ON (retained)
23:17:41 MQTT: sonoff-out/sonoff4/RESULT = {"POWER":"OFF"}
23:17:41 MQTT: sonoff-out/sonoff4/POWER = OFF (retained)
</per>

must have missed something in the docs, just started yesterday night with this repo

HairBear22, 2017-01-07:
I’ll have to double check that when I get home, that would be interesting if that was the issue.

arendst, 2017-01-07:
Several items are in play here when you press the button on Sonoff once.

1. If ButtonTopic = 0 then the button on sonoff always toggles the relay.
2. If ButtonTopic = Topic then the button sends a command to MQTT server with either opposite ON or OFF (as defined with MQTT_STATUS_ON/OFF defines). The MQTT server is supposed to propagate the command back to the sonoff which then would toggle the relay.
3. If ButtonTopic = different from Topic and not 0 it sends a command to MQTT server with either opposite ON or OFF. The MQTT server is supposed to propagate the command to any device listening for the ButtonTopic but it won't be this Sonoff. Another press will therfore send the same command again.

In version 3.1.14 just released it will send in case 3 a TOGGLE message as defined in user_config.h (MQTT_CMND_TOGGLE).

I hope that solves your issue. If not just let  me know.

stoehrmark, 2017-01-07:
Below are the responses i get from the Sonoff. With Option 3 topic set to garagedoor i get back TOGGLE as set in -> #define MQTT_CMND_TOGGLE       "TOGGLE". This is probably the expected response. Can i get back ON and OFF form this custom ButtonTopic instead of  TOGGLE

This is my user_config.h			
#define PROJECT                "iot002" 			
#define MQTT_TOPIC             PROJECT 			
#define MQTT_STATUS_OFF        "OFF" 			
#define MQTT_STATUS_ON         "ON"			
#define MQTT_CMND_TOGGLE       "TOGGLE" 			
			
			
_Below is formatted as follows
Which Input
BUTTONTOPIC
Relay Response
MQTT response_

**OPTION 1**
GPIO0 First Press (onboard button)
0
ON
14:04:43 MQTT: stat/iot002/POWER = ON
~~
GPIO0 Second Press (onboard button)
0
OFF
14:04:43 MQTT: stat/iot002/POWER = OFF
`<<<>>>`
GPIO14 First Press (external button)
0
ON
14:04:43 MQTT: stat/iot002/POWER = ON
~
GPIO14 Second Press (external button)
0
OFF
14:04:43 MQTT: stat/iot002/POWER = OFF

**OPTION 2**
GPIO0 First Press (onboard button)
iot002
ON
14:06:31 MQTT: cmnd/iot002/POWER = ON
14:06:31 MQTT: stat/iot002/POWER = ON
~
GPIO0 Second Press (onboard button)
iot002
OFF
14:06:35 MQTT: cmnd/iot002/POWER = OFF
14:06:35 MQTT: stat/iot002/POWER = OFF
`<<<>>>`
GPIO14 First Press (external button)
iot002
ON
14:06:31 MQTT: cmnd/iot002/POWER = ON
14:06:31 MQTT: stat/iot002/POWER = ON
~
GPIO14 Second Press (external button)
iot002
OFF
14:06:35 MQTT: cmnd/iot002/POWER = OFF
14:06:35 MQTT: stat/iot002/POWER = OFF

**OPTION 3**
GPIO0 First Press (onboard button)
garagedoor
Nothing
14:15:59 MQTT: cmnd/garagedoor/POWER = TOGGLE
~
GPIO0 Second Press (onboard button)
garagedoor
Nothing
14:16:04 MQTT: cmnd/garagedoor/POWER = TOGGLE
`<<<>>>`
GPIO14 First Press (external button)
garagedoor
Nothing
14:15:59 MQTT: cmnd/garagedoor/POWER = TOGGLE
~
GPIO14 Second Press (external button)
garagedoor
Nothing
14:16:04 MQTT: cmnd/garagedoor/POWER = TOGGLE

arendst, 2017-01-07:
Did you set switchmode to 1 in order to get ON/OFF in option 3 for gpio14? 
You'll never get ON/OFF for gpio0 in option3 as the button doesn't know the status of garagedoor as it is supposed to act on unknown status.

stoehrmark, 2017-01-07:
BINGO  that works GPIO14 now responds with On Off

THANKS

closed 2017-01-08 by arendst