arendst/Sonoff-MQTT-OTA-Arduino #207: ButtonTopic with toggle switches

andrewradke, 2017-01-07:
In send_button_power if the state is 2 (toggle relay) MQTT_STATUS_ON is always sent instead.
A quick and dirty fix that has worked for me is:

```
    if (state == 2) {
      snprintf_P(svalue, sizeof(svalue), PSTR("%s"), "2");
//      state = ~(power >> (device -1)) & 0x01; 
    } else {
      snprintf_P(svalue, sizeof(svalue), PSTR("%s"), (state) ? MQTT_STATUS_ON : MQTT_STATUS_OFF);
    }
```

Is there any problems with this or is there a better solution?

arendst, 2017-01-07:
When using ButtonTopic and the name is different from Topic it indeed would only send an ON command.

I've changed it's behaviour when ButtonTopic is different it now sends a TOGGLE command as was suggested here.

Problem was that ButtonTopic toggled related to current relay power state but if the ButtonTopic is different from the Topic it's commands are not meant for this relay. Toggling is then the best option.

andrewradke, 2017-01-08:
Thank you. That was an extremely quick bug fix :-) 👍

closed 2017-01-08 by andrewradke