arendst/Sonoff-MQTT-OTA-Arduino #211: [Feature Request] HomeKit

felixekman, 2017-01-08:
How about adding HomeKit support? There is an amazing project going on here:
https://github.com/HomeACcessoryKid/ESP8266-HomeKit-Demo

I'm imagining MQTT/Alexa/HomeKit co-exisiting with the Sonoff and being amazing!

arendst, 2017-01-08:
Looked at it:
```
Memory: The HomeKit code is approximately 400K and about 18K of RAM is left for other purposes.
```
I have only 450K to play with to keep OTA available on 1M flashes so I am afraid it's a bridge to far for my software, homekit and sonoff.

I suggest to use a home automation tool that both supports sonoff via MQTT (my software) and homekit. That's more flexible too.

felixekman, 2017-01-08:
How about maybe adding a different branch of your git for either W25Q32FV implementations or one without OTA-support?

closed 2017-01-13 by arendst