arendst/Sonoff-MQTT-OTA-Arduino #212: Dual and 4 channel models should enumerate multiple wemos

don-willingham, 2017-01-08:
I have a Sonoff Dual and an Amazon Echo. When I Discover devices in the Echo app, it shows only one WeMo switch, but would be nice if it shows up as two. The 4 channel model would likely discover 4 devices. It would likely be cleaner if Belkin WeMo had multiple channel devices. I am going to try sending multiple UDP packets in wemo_respondToMSearch(), while incrementing the serial/uuid. The functions in webserver will also need some work.

altelch, 2017-01-08:
Don't know if this really works. Depends on how flexible the Echo is with accepting different UDP answers AND using differen setup.xml files. You would also need a friendly name per device. Currently I use the MQTT device name. For multi device support there should be a friendly name per device changeable by MQTT and stored in SPIFFS. A probably cleaner approach would be a HUE-Bridge emulation but that is much more complex than Wemo emulation.

altelch, 2017-01-11:
Here is an Implementation of multiple WeMo Devices on one ESP8266:
https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch

It's rather simple by design but requires a webserver for each emulated wemo device. Not the way I'd prefer.

Greetings, 

                  Heiko.

don-willingham, 2017-01-12:
Until WeMo updates the API, an extra port per extra channel may be necessary. Those additional webservers could be limited to the 3 (I think) URLs specific to the WeMo interface. I'll give it a go when I get more free time.

don-willingham, 2017-01-12:
I've got an ugly hack that needs refactoring, but a second device shows up. I'll clean it up when I get back to it.

https://github.com/don-willingham/Sonoff-MQTT-OTA-Arduino/commit/dc6665c525405c0676cae3145c72fe0b1e55ad97

altelch, 2017-01-12:
I'm looking into a HUE Bridg emulation version for 2 Reasons:

1. Only one webserver
2. Would allow for Dimming (should this ever come in a Sonoff-Device
3. RGB-LED Support

But Hue-Emu ist more complicated. Need to find out how Echo exactly detects the Hue Bridge and how it does access the devices. Hopefully there is just a minimum api setup necessary.

Greetings,
         Heiko.

altelch, 2017-01-15:
Ok, UPnP part seems to work now, Echo requests description.xml and then tries to get the configured lights. Will continue work on http part then.

altelch, 2017-01-16:
Success on HTTP part. Echo now registers a Hue smart home device emulated by the sonoff. The detection process is now understood. Now it's time to implement the required API functions...

altelch, 2017-01-16:
Basic Hue emulation now working. Switching on/off devices now works and every channel is listed as own device on virtual HUE bridge.

altelch, 2017-01-19:
Hi Don,

I used your code to provide a HUE Emulation mode with 4 different friendly names. So we now have a multi channel solution without different webservers and completely different names per channel. You may try my repository if you want to try it.

Greetings,

             Heiko.