arendst/Sonoff-MQTT-OTA-Arduino #214: How to update firmware OTA?

JoeTN, 2017-01-08:
Hello Arendst
I have used your great firmware on my Sonoff's for quite some time now, and it just works great!
- This is probably dead simple, so excuse me for asking the question, but how do I upgrade the firmware OTA? - It's currently running  V2.0.13, and I can point it to the firmware file(s) by using the webinterface, but I can't read anywhere, which file I should point it to? - I believe I have to find it somewhere in the arduino or sonoff folder at your repository, but I simply can't...

I hope you are willing to help me.

Regards
Joe

arendst, 2017-01-08:
Joe,

glad it works for you.

If you want to upgrade a Sonoff Basic then you can just upload the latest binary from folder api/arduino.

If you use Pow or other features you'll have to recompile as documented in the wiki.

JoeTN, 2017-01-08:
Aaaaah :-)
- I am using a Sonoff Basic, when I try to upload sonoff.ino.bin it fails with this error: File magic header does not start with 0xE9

Can I fix that somehow?

JoeTN, 2017-01-08:
Never mind.. I just redownloaded the bin file, now it works :-)

closed 2017-01-08 by arendst