arendst/Sonoff-MQTT-OTA-Arduino #220: Sonoff Touch wont flash

connectedthings, 2017-01-10:
Apologies upfront for posting this here (have joined the Google group am pending but am highly frustrated hence my post :( ) 

I have been struggling to flash a Sonoff Touch ... yes I have read the notes etc. am using external power supply (have tried several) , common ground and only RX/TX to USB (once again tried different converter chip sets). I have used this setup since day dot and never had a problem with Generic ESP12's, Sonoff and S20

Without GPIO0 grounded aka regular boot serial monitor shows bootloader announcement. Once I ground GPIO0 things get weird .... if I remove GPIO0 wire after boot serial receive on USB blinks erratically

When I try flash (have selected Generic8285) on upload start send flashes a couple of times and then

trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

Any ideas before I jump off a building!

Tx folks

And as always, Theo, tx for all the great work ... lots of activity on Domoticz, will have to go give it a try ...

jaysuk, 2017-01-10:
Please see this other github repository with details about how to flash to sonoff touch
https://github.com/davidmpye/Sonoff-Touch-MQTT

connectedthings, 2017-01-10:
Tx jaysuk am aware of that project and its also not quite correct ...

"4) Using either the Arduino IDE or platform.io, upload the firmware. (If using arduino, 'Generic ESP8266 board' and Flash: (1M, 64K SPIFFS). If you don't get that part set correctly, it is likely OTA Upgrades will fail in future."

The Sonoff Touch is based on ESP8285 which has onboard FLASH and needs to be flashed in DOUT mode!

jaysuk, 2017-01-10:
It looks as though you may need to hold GPIO0 low through the whole of the flashing process.
[`](https://uploads.disquscdn.com/images/698f28c20f5ebcd22bf839ee7cb5db6457ca31a3f5c6b1f0bbb74823792a2811.jpg)

arendst, 2017-01-10:
See wiki Compile and upload. ESP8285 and dout are important.

connectedthings, 2017-01-10:
Tx Theo, have done all that, still getting same result, will keep trying, have also pinged ITEAD, might be a faulty unit ...

closed 2017-01-15 by arendst

TRIROG, 2017-05-16:
Any news ragerding this? i seem to also have same issue?

Upon flashing i get 
`Fatal exception (0): 
epc1=0x40217e08, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
`
in serial

chimeranzl, 2017-07-31:
warning: espcomm_sync failed often means you have your RX/TX wires around the wrong way

Naobana, 2018-01-21:
Hi connectedthings,

Have you fixt this problem? Ingaven exaxtly the same errors and i dont know howto fix it.

Thanks!

AlexTransit, 2018-01-22:
Use this
[https://github.com/davidmpye/Sonoff-Touch-MQTT](url)

connectedthings, 2018-01-22:
Yes. I did solve it, very simple issue in my case. Make sure nothing is touching the touch pad while you are trying to flash, if the touch pad activates in anyway during this process it will not work

Naobana, 2018-01-22:
Great, will try that. Thanks

mrohner, 2018-01-26:
"The Sonoff Touch is based on ESP8285 which has onboard FLASH and needs to be flashed in DOUT mode!"

How to do that?

mrohner, 2018-01-26:
@AlexTransit Your link does not work 404

AlexTransit, 2018-01-27:
sorry
[https://github.com/arendst/Sonoff-Tasmota](url)

Naobana, 2018-01-28:
I have also used Sonoff-Tasmota. Flashing via the serial was not a great success... tried everything but no positive results. Tasmota fixed it in 2 minutes!