arendst/Sonoff-MQTT-OTA-Arduino #222: Feature request: BH1750 Light Sensor

RobKuipers, 2017-01-10:
BH1750 is a digital Ambient Light Sensor IC for I2C bus interface. 
I use this on an earlier version of Sonoff-MQTT-OTA-Arduino (1.0.11) to report to Node-Red (via a topic 'LIGHT'), triggering a flow that -when dark- switches a scene in Domoticz (which in turn switches -amongst others- the relay of the very same Sonoff :-) ). Works nicely.

Would this be feasible as a feature for this firmware?

Eg. something along the following lines:

`#ifdef SEND_TELEMETRY_BH1750
/*********************************************************************************************\
 * BH1750 GY-302  Ambient Light Sensor
\*********************************************************************************************/

#define BH1750_ADDR         0x23

#define BH1750_CONTINUOUS_HIGH_RES_MODE   0x10  // Start measurement at 1lx resolution. Measurement time is approx 120ms.

float bh1750_readLux()
{
    Wire.requestFrom(BH1750_ADDR, 2);
    byte b1 = Wire.read();
    byte b2 = Wire.read();
    float val=0;     
    val=((b1<<8)|b2)/1.2;
    String log = F("BH1750  : Light intensity: ");
    log += val;
    addLog(LOG_LEVEL_INFO,log);
    return val;
}

bool bh_init()
{
  Wire.beginTransmission(BH1750_ADDR);
  Wire.write(BH1750_CONTINUOUS_HIGH_RES_MODE);
  error = Wire.endTransmission();
  return (error==0);
}

uint8_t bh_address()
{
  return (uint8_t)BH1750_ADDR;
}

uint8_t bh_detect()
{
  return bh_init();
}
#endif  // SEND_TELEMETRY_BH1750`

arendst, 2017-01-11:
Implemented in version 3.2.0 just released

RobKuipers, 2017-01-13:
Wonderful. This firmware is like a Swiss army knife: it can solve just about every use case I throw at it!
Thanks again Theo.

JoeWiseman, 2017-01-29:
Hey Rob and Theo,

I've red about using an brightness sensor alon with the Sonoff-devices. As I'm not familiar with all this arduion stuff I would like to ask two things:

Rob: can you tell me along with which device you did use the BH1750, and of course HOW did you use it?

Theo: you might remember that I'm the one who is using the sonoff devices without MQTT, so just along with the http-requests. So is it possible to get a numerical value of the measured brightness as a response  th a http-command?

You be great to get some answers from you!

Regards
Joe

arendst, 2017-01-30:
Joe,

via this link https://github.com/arendst/Sonoff-Tasmota you'll find my latest version of this software which allows for easy connection and configuration of sensors.

Requesting ad-hoc sensor data is possible with command ```status 10```. Also via http as ```http://sonoff/cm?cmnd=status%2010```.

Whith a HTU21 and a BH1750 ...

[image]

...connected to your configured I2C pins defined on the Module Configuration web page...

[image]

...the resulting in message would be
 ```
RESULT = {"StatusSNS":{"Time":"2017-01-30T10:47:15", "HTU21":{"Temperature":"21.7", "Humidity":"41.0"}, "BH1750":{"Illuminance":95}}
```

JoeWiseman, 2017-01-30:
Thanks for the hints .... Unfortunately I seem to be to stupid to get the sensor connected without a conrete plan, photo or tutorial ... Maybe you can give a little hint where to find how to "connected to your configured I2C pins defined on the Module Configuration web page".

As I understand your answer and the pictures above it ispossible to use a sensor (combination of HTU21 and a BH1750) along with the Sonoff-basic-device.
Is there any way to use a single BH1750 along with a Sonoff TH, which seems to be layed out for sensor usage?

About the software I have a question too: I guess that the Sonoff-Tasmota-software is esecially for egtting sensors connected. To me it seem to be based on the software in this section, so is it possible to switch off the MQTT as well (via user_config.h)?

Thanks for answer in advance!

Dietmar-Franken, 2017-01-30:
@JoeWiseman 

Look at line 37-38 in user_config.h

// -- MQTT ----------------------------------------
#define MQTT_USE               1                 // [Mqtt] Select default MQTT use (0 = Off, 1 = On)

JoeWiseman, 2017-01-30:
@Dietmar-Franken:
Thanks [Danke ;-)] for the hint, Dietmar. That was the easy part ...
Did you ever use the BH1750 along with a Sonoff-basic device?

RobKuipers, 2017-01-30:
Hi Joe,

I'm using this 1750 sensor on a regular Sonoff. Can't help with the exact pin connections I used at the moment , as I am abroad. But I can asure it works!
I'll be back home next week, so please let me know if you still need help then.

Regards, Rob

JoeWiseman, 2017-01-31:
Hi Rob,

thanks for answering. I would be very pleased, if you can give me your pin connections! The only thing I've found was a DHT22 sensor, but this one has only got 3 cables to be connected:
1 ---------- 1 (VCC)
2 ---------- 5 (GPIO14)
3
4 ---------- 4 (GND)
DHT22  / Pins on Sonoff

But after all I'm not in a hurry, so I can wait .... 
Regards, Joe

RobKuipers, 2017-02-05:
Hi,

home again. Just took a look. I've used:
  #define I2C_SDA_PIN          3            
  #define I2C_SCL_PIN          1            
Does that help?

JoeWiseman, 2017-02-05:
Hi Rob,

thanks for answering!
Assuming that we both talk about the row of 5 pins which are used, when programming I found that unfortunately there are serveral different ways of numbering the pins in the internet ....

To be totally right I would be glad if you ...
... can give me your full pin assignment (I think the BH1750 has got the following pins: VCC, GND, SCL, SDA, ADDR) 

and
... would be kind to tell me wether your numbers (1...5) are going from the upper ("quadratic") pin to the lower ("round") one or from the lower pin to the upper one?
[image]

Thanks for helping me out in advace!

RobKuipers, 2017-02-07:
joe, I build the whole thing into my cooker hood (afzuigkap)...

But I found this website that explains the Sonoff pins quite well: http://www.andremiller.net/content/programming-an-itead-sonoff-wireless-smart-switch-esp8266

And as I remember, the ADR line on the BH1750 can just be left open. Plenty of pictures of its pin layout on the sellers websites.

JoeWiseman, 2017-02-07:
Rob, thanks for the post.
I think that this brought some light into the darkness ;-)
I'll try as soon as the sensor will arrive!
It's good to have someone helping hand on the web! I don't know what I would have done in the digital Stone Age in the last century :-D
Thanks again!!