arendst/Sonoff-MQTT-OTA-Arduino #224: Alexa compatible

danjperron, 2017-01-11:
How do you make it compatible with echo dot?   Domoticz using  HA-Bridge? 

On my part I use fauxmo on a Raspberry Pi which emulates the Wemo for MQTT device.

https://dl.dropboxusercontent.com/u/48891705/chip/AlexaKitchen.MP4

altelch, 2017-01-11:
Hi,

the sonoff itselfe acts like a belkin device. Reacts on the UDP search packet, sends the fitting answer and has the same web urls like a wemo. So the Echo can find it and thinks it's a wemo device.

      Heiko.

JoeTN, 2017-01-11:
I hope it's okay that I "borrow" your thread. I have a question about the Amazon Echo support, and if it could be answered here, there's no need for me to start a new thread.

I just uploaded firmware v.3.1.16 to a new Sonoff Basic through Arduino IDE. The only thing I did to enable Alexa support, was to enable line 106, #define USE_WEMO_EMULATION. Then I uploaded the firmware and connected the relay to my wifi with success. Afterwards I've set a new friendly name as testrelay. My problem is, that I can't get my Echo Dot to register the device. Normally it registers devices just fine on my network. Do I have to enable any additional settings or?

I tried configuring the web log level to 3, to catch information in the console about the problem (I don't even know if I'm supposed to do it that way?) but anyway I don't see anything there when discovering devices from Alexa.

Regards
Joe

altelch, 2017-01-11:
Nope, no additional Settings are required. Check support.ino. There should be a commented line with a debug print output to see when a UDP packet is received in pollUDP(). You can also try to load http://<IP>/setup.xml to see if the http part works. Check the log for a Multicast (re)connect message.

Greetings,
             Heiko.

JoeTN, 2017-01-11:
Weird.. I've enabled line 673 addLog_P(LOG_LEVEL_DEBUG, packetBuffer); in pollUDP() as I expect that's the one you were referring to. I reuploaded the firmware by using Arduino IDE, expecting to see new stuff about the UDP packets in the console, but I didn't see anything new. I am supposed to look for the debug info in the console output, right?
- I've checked setup.xml, it looks like this:

<root>
<device>
<deviceType>urn:Belkin:device:controllee:1</deviceType>
<friendlyName>myrelay</friendlyName>
<manufacturer>Belkin International Inc.</manufacturer>
<modelName>Sonoff Socket</modelName>
<modelNumber>3.1415</modelNumber>
<UDN>uuid:Socket-1_0-201612K0517980</UDN>
<serialNumber>201612K0517980</serialNumber>
<binaryState>0</binaryState>
<serviceList>
<service>
<serviceType>urn:Belkin:service:basicevent:1</serviceType>
<serviceId>urn:Belkin:serviceId:basicevent1</serviceId>
<controlURL>/upnp/control/basicevent1</controlURL>
<eventSubURL>/upnp/event/basicevent1</eventSubURL>
<SCPDURL>/eventservice.xml</SCPDURL>
</service>
</serviceList>
</device>
</root>

I am appreciating your help, Heiko :-)

altelch, 2017-01-11:
Hi,

the Multicast (Re)Joined is printet in Informational loglevel. The received UDP Packets in DEBUG. So if your Weblog is in Debug level you should see something when attached to the log. Do you have other Wemo-Devices? Hue for example uses a bit different mechanisms. Maybe there is a Problem with Multicast-Support on your WLAN-AP(s). Normaly there is enouth noise on SSDP to see some packets.

Greetings,

            Heiko.

JoeTN, 2017-01-11:
I believe it is in debug as I set web log level to "3 Debug" through the webinterface. But I'm not seeing any packets in the console. sonoff-ip-here/cs. I don't have any other Wemo devices. I'm emulating a Hue-bridge made by BWS Systems, that's all.

It could be a problem with multicast-traffic. I'm having a regular plain consumer-router which also functions as WAP. It works with any other device such as my Chromecast eg.

I'll try to investigate it further, it seems to be a problem on my network.

JoeTN, 2017-01-11:
Btw. This is the console output after restarting the relay
`00:00:00 Config: Loaded configuration from flash at F8 and count 19
00:00:00 APP: Bootcount 10
00:00:00 APP: Project sonoff (Topic sonoff, Fallback DVES_07E75C, GroupTopic sonoffs) Version 3.1.16
00:00:00 Config: Saved configuration (928 bytes) to flash at F7 and count 20
00:00:00 Wifi: Attempting connection...
00:00:00 Wifi: Patch issue 2186
00:00:00 Wifi: Connecting to AP1 EbbesensResidence in mode 11N as sonoff-1884...
00:00:01 Wifi: Attempting connection...
00:00:02 Wifi: Attempting connection...
00:00:03 Wifi: Attempting connection...
00:00:04 Wifi: Connected
00:00:04 mDNS: Initialized
00:00:04 HTTP: Webserver active on sonoff-1884.local with IP address 192.168.1.110
00:00:04 UPnP: Multicast (re)joined
00:00:06 HTTP: Handle root
00:00:07 HTTP: Handle console
00:00:07 MQTT: Attempting connection...
00:00:07 RTC: (UTC) Wed Jan 11 19:01:43 2017
00:00:07 RTC: (DST) Sun Mar 26 02:00:00 2017
00:00:07 RTC: (STD) Sun Oct 29 03:00:00 2017
20:01:44 mDNS: Query done with 0 mqtt services found
20:01:58 MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds`

It says multicast (re)joined.

altelch, 2017-01-11:
Then you should see an Hue-Bridge announcement about every 20 seconds. The emulator does this, too if I remember correctly. But I have to admin that this is strange if the emulated Hue works as the echo device also uses UDP/Multicast to detect the bridge. Do you have a working MQTT server? The sonoff fails to connect.

JoeTN, 2017-01-11:
I don't see any announcements from the Hue Bridge, which is as you say weird if they were supposed to show up also. The Hue bridge works just fine. I have a working MQTT server yes, I'm currently using it for one of my other relays only, so I haven't configured this relay to use the MQTT server, as my idea was to just have Alexa discovering the relays itself, so that I could avoid using the MQTT server.

JoeTN, 2017-01-11:
Just to ensure that we are on the same page, this is how my logging is configured: http://imgur.com/a/TGFQ8 and this is what I see when discovering devices: http://imgur.com/a/LuMtu

altelch, 2017-01-11:
Please try a smartphone with the WeMo App (won't work with the sonoff-code) or the Hue App and initiate a device / bridge search. If you still won't get any UDP packets while running the search you have a problem with multicast. The HueBridge is probably detectet by other means...there are several ways for Hue devices.

Sorry,

                Heiko.

altelch, 2017-01-12:
Maybe this is somehow related to #228. Compiled 3.2.0 yesterday evening to test this and had the same behaviour. No UDP packets where received. The 3.1.16 I used before worked as expected.

Greetings,

            Heiko.

JoeTN, 2017-01-12:
I first tried the Wemo-app, but as far as I can see, I can't "discover" anything form there, it just tells me to connect to the hotspot created by the wemo-unit and then launch the Wemo-app to go on from there, since the Sonoff-relay isn't creating a hotspot or anything, I hit the wall there.. Am I missing something obvious here?

I then tried the Hue app, which can search for bridges, while searching for bridges I didn't see anything in the console of my sonoff-relay, which is still configured to show level 3 debug info.

---- From now on I'm on deep water -----
I have very little experience in Wireshark, so I tried applying the following filter to the traffic stream: ip.addr >= 224.0.0.0 && ip.addr == 192.168.1.110 - it should supposedly show all multicast-traffic, in this case regarding my sonoff-relay since it has ip 192.168.1.110. It gives me the following output while "idle": http://imgur.com/a/ohfKy - I tried asking Alexa do discover devices which didn't change anything.

I tried applying the same filter, but with the ip of my phone, while searching for a bridge through the Hue-app which gave this: http://imgur.com/a/NiTut

- As I said, I'm on deep water with this Wireshark info, I have no idea whether it's usable or not...

altelch, 2017-01-12:
Hi, as written above there might be a Problem with 3.2.0. Please check out the 3.1.16 again and try this version first. For Wireshark you should just use "host" 239.255.255.250 or protocol UDP and port 1900.
There are the multicast packets of the SSDP services.

Greetings,

            Heiko.

JoeTN, 2017-01-12:
I am currently running 3.1.16, (Downloaded here: https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/tree/master/sonoff) this is with this version that I do not receive information about any UDP packets in the console of the sonoff-relay.

Using this filter to filter SSDP from the Sonoff, gives nothing: ip.addr == 239.255.255.250 && ip.addr == 192.168.1.107. 

While trying it with the IP of my phone while discovering Hue bridges gives this: http://imgur.com/a/HPqyz

altelch, 2017-01-12:
Hi, tried it again with 3.1.16 and get lot of UDP packets from my echo. So the code works. There must be a Problem with your WLAN/AP Setup or config which prevents Multicast traffic to reach your sonoff.

Sorry,
          Heiko.

altelch, 2017-01-12:
Here the console log of multicast traffic on my sonoff:

21:30:20 M-SEARCH *
21:30:20 UPnP: Response sent to 192.168.100.30:4102
21:30:21 HTTP: Handle Hue Bridge setup
21:30:24 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:schemas-upnp-org:device:basic:1

21:30:24 UPnP: Response sent to 192.168.100.38:50000
21:30:24 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:schemas-upnp-org:device:basic:1

21:30:24 UPnP: Response sent to 192.168.100.38:50000
21:30:24 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:Belkin:device:**

21:30:24 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:Belkin:device:**

21:30:25 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:Belkin:device:**

21:30:25 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:Belkin:device:**

21:30:25 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:schemas-upnp-org:device:basic:1

21:30:25 UPnP: Response sent to 192.168.100.37:50000
21:30:25 M-SEARCH * HTTP/1.1
HOST: 239.255.255.250:1900
MAN: "ssdp:discover"
MX: 15
ST: urn:schemas-upnp-org:device:basic:1

My code currently contains first code for HUE Bridge Emulation. The echo accepted the UDP response and startet an HTTP query for the service description xml.

Greetings,

          Heiko.

samsixtysix, 2017-01-12:
>  I haven't configured this relay to use the MQTT server

My echo dot would not discover any devices that were not connected to a running MQTT broker.  Once detected they could be controlled without the broker running.

altelch, 2017-01-12:
Ok, it's mqtt related. You need a working mttq connection to get UDP running. Else during every mqtt_reconnect the udp connected flag is reset. So disable MQTT in user_config.h or configure a server.

Greetings,

          Heiko.

citrusfizz, 2017-01-13:
I'm loving this custom firmware.  Alexa was easily able to pick up the device and control it once I had an MQTT server setup and the Sonoff connected to it successfully, and of course the line 106 un-commented before flash.     Took a couple tries to get this right. 

I understand that the MQTT uses the mDSN and resets if unsuccessful. That being said, do you think there could be away around this in future updates to the code?  I'd like to use this firmware somewhere that doesn't have a MQTT server.   I know that is kinda the point of this firmware (MQTT)  but it'd be nice if it wasn't a requirement for Alexa/Wemo to work.

altelch, 2017-01-13:
Check user_config.h and comment out #define USE_MQTT. That should solve this.

Greetings,

               Heiko.

danjperron, 2017-01-13:
Thanks a lot for everything.

It works directly with Alexa  but I prefer fauxmo with MQTT for some obvious reason.
One point of connection (firewall), Multiple vocal response, Be able to add code to control something else like GPIO of my raspberry Pi.

If you want to use fauxmo this is what I used for script. I create my own class to simply add sonoff device with a table of vocal sentence and a table of MQTT topic.

https://dl.dropboxusercontent.com/u/48891705/chip/home2.py

I use a Raspberry PI A for MQTT and fauxmo script.

closed 2017-01-13 by arendst

JoeTN, 2017-01-13:
Okay, so I tried connecting the relay to my MQTT-server, and it immediately showed UDP packets in the console, just like the ones you posted, Heiko. - Now I just have to get my Echo to discover it, which it still doesn't - but I guess it's a matter of tinkering a bit with it. At least It's going in the right direction now!

It would make sense (in my non-programming head at least...), if it could work without the MQTT-connection, since the Alexa-support would allow me to stop using the MQTT-server. As far as I understand from your replies, I can simply just disable the MQTT support (//#define USE_MQTT). But would I then be able to activate it again on a later point, from the console if I implement the relay in a place where it can't easily be accessed, or is the only way to reflash the relay to enable the MQTT again?

As I said, I really appreciate all your help in helping me with my problem! :-)

arendst, 2017-01-13:
The USE_MQTT define throws all MQTT code overboard as that seems to be wanted by the Alexa people.

If you want MQTT again you'll have to recompile with the USE_MQTT flag enabled. Then You are finally back in my courtyard ;-).

JoeTN, 2017-01-13:
Okay, but if I disable the USE_MQTT in the firmware, Alexa won't be able to discover the relay, right?
- As I'm getting it now, the Alexa-support relies on that the USE_MQTT is activated, because it won't enable "UDP-traffic" otherwise

arendst, 2017-01-13:
Yes you're right. That's a bug I have to solve.

seteq, 2017-05-10:
Has this ever been solved? Is a MQTT server still needed for the firmware to work with alexa?