arendst/Sonoff-MQTT-OTA-Arduino #225: cant connect to aio

jesusvallejo, 2017-01-11:
i cant connect to adafruit io´s mqtt server , i think it´s related with the mqtt finguerprint , i did´t edited it cause i could not find it on any mqtt conf on the internet.

these are the promted lines:
00:00:00 Wifi: Connecting to AP1 WLAN_2DB2 in mode 11N as sonoff-6394...
00:00:04 Wifi: Connected
00:00:04 mDNS: Initialized
00:00:04 HTTP: Webserver active on sonoff-6394.local with IP address 192.168.1.41
00:00:05 MQTT: Attempting connection...
15:38:03 mDNS: Query done with 0 mqtt services found

mverzakov, 2017-01-13:
I have the same problem with my local mosquitto mqtt server. 
Also, I've tried to connect to cloudmqtt but it was unsuccessful.

jesusvallejo, 2017-01-13:
ive tryed also cloudmqtt with the same result , i keep thinking it has something to do with mDNS

jesusvallejo, 2017-01-13:
is not like i "suspect" (sorry if that annoyed you , it was not my aim) , i was kind of asking... could it be caused by mDNS

mverzakov, 2017-01-14:
I've tried that as well. And all I got is:
_04:30:43 MQTT: Attempting connection..._

jesusvallejo, 2017-01-14:
i could connect them to mosquitto on a rpi3 , but the commands does not work the console shows this every time i publish a topic like: "cmnd/sonoff/Light on"

it shows this:
`17:01:30 MQTT: stat/sonoff/COMMANDS1 = Status, SaveData, SaveSate, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname
17:01:30 MQTT: stat/sonoff/COMMANDS2 = MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, PowerRetain, MqttUnits, MessageFormat, GroupTopic, Timezone, LedState, TelePeriod
17:01:30 MQTT: stat/sonoff/COMMANDS3 = Power, Light, PowerOnState, PulseTime, BlinkTime, BlinkCount, Weblog, Webserver, DomoticzInTopic, DomoticzOutTopic, DomoticzIdx, DomoticzKeyIdx, DomoticzUpdateTimer
17:02:30 MQTT: tele/sonoff/UPTIME = 1`

jesusvallejo, 2017-01-14:
commands like : cmnd/sonoff1/Power  are recognised
but commands like: cmnd/sonoff1/Power 1 arent, and shows what i posted above .
why could  be causing this???

arendst, 2017-01-14:
See wiki on top of this page in section MQTT for Newbies regarding message and payload

mverzakov, 2017-01-16:
Well, it's definitely something wrong with mDNS. 
My MQTT broker is on the RPi3 with local DNS. 
So I tried to connect Sonoff by hostname _raspberry.local_ and it wasn't worked.
Then I set up static IP for raspberry and change MQTT host to IP address instead of hostname and it works.

arendst, 2017-01-16:
For mDNS issues see #115.

closed 2017-01-24 by arendst