arendst/Sonoff-MQTT-OTA-Arduino #227: features request

aferraz, 2017-01-11:
Hi,
 first of all my congratulations for your great work.
If I can I would like to suggest a features that could be interesting and useful:
New message at midnight with last day Power consuption, It can be used to store consuption data in a DB

Thank You
 Alessandro

arendst, 2017-01-11:
Will do

aferraz, 2017-01-11:
Sorry, what about an alarm when the unit reaches its Ampere or Watt limit for more that n minutes ? Also a auto Power OFF function in this conditions could  avoid damages

Thanks

arendst, 2017-01-13:
Looked at it and I think all your requests are already implemented:
- Yesterday Energy usage is send with every Telemetry update. Just set TelePeriod to 300 and you'll receive the requested data.
- Alarms for PowerLow/High, CurrentLow/High and VoltageLow/High are also available see the wiki and command list. Your home automation tool could trigger and act upon the information
- Power limits are rudimentary available but still need to be investigated. You could give it a try by enabling define FEATURE_POWER_LIMIT around line 51 in sonoff.ino

aferraz, 2017-01-13:
Perfect, I give it a look. Thanks

closed 2017-01-15 by arendst

aferraz, 2017-01-19:
I tried what you wrote but I do not understand how it works.
I have set:
#if MODULE == SONOFF_POW
  #define FEATURE_POWER_LIMIT
#endif
#define MAX_POWER_HOLD         10           // Time in SECONDS to allow max agreed power
#define MAX_POWER_WINDOW       30           // Time in SECONDS to disable allow max agreed power
#define SAFE_POWER_HOLD        10           // Time in SECONDS to allow max unit safe power
#define SAFE_POWER_WINDOW      30           // Time in MINUTES to disable allow max unit safe power
#define MAX_POWER_RETRY        5            // Retry count allowing agreed power limit overflow

and then from the console:
21:12:02 MQTT: stat/sonoff4/RESULT = {"PowerHigh":"10"}
21:12:02 MQTT: tele/sonoff4/TELEMETRY = {"PowerHigh":"ON"}
21:12:07 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:12:07", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"TP-LINK_0DB6", "RSSI":84}}
21:12:07 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:12:07", "Energy":{"Yesterday":"4.081", "Today":"0.494", "Period":0, "Power":12, "Factor":"0.55", "Voltage":225, "Current":"0.101"}}
[..]
21:15:09 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:15:08", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"TP-LINK_0DB6", "RSSI":84}}
21:15:09 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:15:09", "Energy":{"Yesterday":"4.081", "Today":"0.494", "Period":0, "Power":12, "Factor":"0.54", "Voltage":225, "Current":"0.103"}}
21:15:13 MQTT: stat/sonoff4/RESULT = {"PowerHigh":"10"}
21:16:04 MQTT: stat/sonoff4/RESULT = {"PowerHigh":"1"}
21:16:12 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:16:12", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"TP-LINK_0DB6", "RSSI":86}}
21:16:12 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:16:12", "Energy":{"Yesterday":"4.081", "Today":"0.495", "Period":0, "Power":12, "Factor":"0.56", "Voltage":224, "Current":"0.098"}}
21:16:17 MQTT: stat/sonoff4/RESULT = {"CurrentHigh":"1"}
21:16:18 MQTT: tele/sonoff4/TELEMETRY = {"CurrentHigh":"ON"}
21:17:19 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:17:19", "Uptime":0, "POWER":"ON", "Wifi":{"AP":1, "SSID":"TP-LINK_0DB6", "RSSI":88}}
21:17:19 MQTT: tele/sonoff4/TELEMETRY = {"Time":"2017-01-19T21:17:19", "Energy":{"Yesterday":"4.081", "Today":"0.495", "Period":0, "Power":12, "Factor":"0.55", "Voltage":224, "Current":"0.102"}}
[..]

I would expect a warning message, but nothing happens. Something wrong?

Thanks

Alessandro

arendst, 2017-01-19:
You are mixing up two seperate features.
Forget about power limits for the moment and focus on Alarms.

See wiki commands about PowerHigh and PowerLow. Suppose your constant load is 100W. If you set PowerHigh to 120 you will receive a message when the load goes over 120W. Same forVoltage and Current. It is only alarm and nothing is switched off by Pow. Your home automation tool needs to act on the alarms.

In the case of power limits a window is opened where a total amount of energy might be used. If the amount is reached power is turned of by Pow. This does not use the previous powerhigh limits. Another options allows for a max of peak power for 10 seconds where pow will power off, wait for30 seconds and then reapply power in sequence of 5 after which power is turned off. This is the feature power limit that still needs testing.

aferraz, 2017-01-19:
Hi arendst,
 Ok for the PowerHigh limit, I had got the hang, but if you look at my log there isn't any alarm message. On the sonoff I have 12W and set PowerHigh first to 10, after to 1, but no alarms.

In the second case, power limits, where I should set this limits ?

Thanks

Alessandro

arendst, 2017-01-19:
The second line is your alarm at 21:12:02 as your power is 12 and your high is 10.

For the use of power limits look at the source for the defines of this feature.

aferraz, 2017-01-19:
OK, thanks