arendst/Sonoff-MQTT-OTA-Arduino #228: Problem: MQTT commands no longer work in 3.2.0

tam481, 2017-01-11:
Hi Theo,
I've upgraded to version 3.2.0 and it looks like the subtopic has been changed to POWER which is fine I can change that in the override file. However, MQTT commands seem to have stopped working. For example, 

`mosquitto_pub -t cmnd/home/livingroom/TableLamp/POWER -m OFF`

generates the following entries in the console

```
MQTT: stat/home/livingroom/TableLamp/COMMANDS1 = Status, SaveData, SaveSate, Upgrade, Otaurl, Restart, Reset, WifiConfig, Seriallog, Syslog, LogHost, LogPort, SSId1, SSId2, Password1, Password2, AP, Hostname
MQTT: stat/home/livingroom/TableLamp/COMMANDS2 = MqttHost, MqttPort, MqttUser, MqttPassword, MqttClient, Topic, ButtonTopic, ButtonRetain, PowerRetain, MqttUnits, MessageFormat, GroupTopic, Timezone, LedState, TelePeriod
MQTT: stat/home/livingroom/TableLamp/COMMANDS3 = Power, Light, PowerOnState, PulseTime, BlinkTime, BlinkCount, Weblog, Webserver
```

I've tried resetting using the console by entering "Reset 2" 

I've also noticed that entering POWER OFF or ON in the console doesn't work either

Free memory is also showing as 22k as opposed to 29/30 before 3.2.0

Hogesyx, 2017-01-12:
Can you test with non / topic name and see if it works? eg home_livingrom_tablelamp.

danjperron, 2017-01-12:
I got the same thing. The sonoff switch published  but can't received any subscribe stuff.

http://imgur.com/a/9VItm

Every time I publish cmd/sonoff_salon1/POWER, ON I got the COMMANDS1..S3 stuff. 

P.S. I tried just salon1 for topic and it didn't work.

This is my first trial on V3.2 since all the other switch works well with 2.1.1

Hogesyx, 2017-01-12:
I haven't test any post 3.1.13, not sure if this is due to the #152 fix.

arendst, 2017-01-12:
That explains why my letterbox alert didn't work this morning ;-)

Sorry guys. Just revert to 3.1.16 until further notice.

tam481, 2017-01-12:
I can confirm that reverting to 3.1.16 resolves the issue 👍

arendst, 2017-01-13:
Should be OK now with version 3.2.1 just released (albeit Friday 13th ...)

closed 2017-01-13 by arendst