arendst/Sonoff-MQTT-OTA-Arduino #231: flashed touch does not work for me

DerKorb, 2017-01-12:
I flashed several sonoff basic and pow so far without any problem using platform.io
I also flashed two sonoff touch successfully (data is written and the final success message appears), but in the end, when i power the touch nothing happens. No led, no dhcp client appearing like with the others. I tried one of the touches before flashing them and it did work with the original firmware. Did this work for everyone else? Is the led supposed to be blinking when i connect power?

danjperron, 2017-01-12:
Did you select the correct cpu  esp8265?

DerKorb, 2017-01-12:
no, i thought i saw somewhere in the platform.io file that it was done automatically. but when i tried to relocate that part i could not find it anymore. guess you are right then and this is my problem. will try tomorrow.

DerKorb, 2017-01-13:
That fixed it. For anyone else wondering how it is done when using atom/platformio.io: you need to change board in platformio.ini to esp8285 from esp01_1m.

closed 2017-01-13 by DerKorb