arendst/Sonoff-MQTT-OTA-Arduino #232: Toggle Button on Root Webpage not working

citrusfizz, 2017-01-13:
I have three Sonoff Basics and they were all using verion 2.1.2.  I notice that in both Chrome and Firefox that the Toggle button didn't seem to do anything at all.  I figured it'd probably be fixed in another version. 

Today i saw 3.1.16 and flashed them all to that as you now have support for Alexa. 
However, again, this toggle button seems to do nothing.   

I turned on debug (console)   and viewed the Chrome Console as well when i press this button.  But it's as if it really doesn't do anything.  No change in my devices, console or anything when i press that button. 

Am i missing something?

arendst, 2017-01-13:
No idea why. Works for me in Chrome, Firefox, IE, Edge...
```
11:56:33 RSLT: DataCb Topic wemos5, Group 0, Index 1, Type WEBLOG, Data 3 (3)
11:56:33 MQTT: stat/wemos5/RESULT = {"WebLog":3}
11:56:36 HTTP: Handle root
11:56:38 HTTP: Handle root
11:56:38 MQTT: stat/wemos5/RESULT = {"POWER":"ON"}
11:56:38 MQTT: stat/wemos5/POWER = ON
11:56:39 HTTP: Handle root
11:56:39 MQTT: stat/wemos5/RESULT = {"POWER":"OFF"}
11:56:39 MQTT: stat/wemos5/POWER = OFF
11:56:40 HTTP: Handle console
```

crankyoldgit, 2017-01-17:
FWIW, I've had the Toggle button not work occasionally after restart, until pressed a couple of times. Then it works. This may be due to connectivity issues if the processor is currently stuck in a network operation, it may cause the UI to appear non-functional. e.g. unable to connect to a MQTT server.

Also, I've seen this sort of issue with sick/bad Sonoff devices. i have 2 out of 19 sonoffs that behave poorly. One has very poor signal strength (even when next to the AP) and times out large packets. The other has great signal (RSSI > 90) but seems to crap out when sending large packets. Both of which exhibit the symptoms you are seeing too. Thus you may possibly have bad devices.

smadds, 2017-01-17:
If the MQTT setting are not correct (e.g. if left with the author's default settings) there is a relatively long period of being unresponsive while waiting for a login timeout. Could this be affecting you?

citrusfizz, 2017-01-20:
i just tested a sonoff that has been on for a while,  still doesn't seem to do anything.  i did see it work (once)
i control the device with my MQTT server so i know that is setup correctly.

btsimonh, 2017-01-24:
I'm running 3.1.8 on 3 devices (2 sonoff, 1 custom), and toggle from the web interface has never worked for me (chrome on windows).  About to flash a slampher later tonight; will try latest and report.