arendst/Sonoff-MQTT-OTA-Arduino #236: POW triggered telemetry data?

Reinhartd, 2017-01-13:
Hello, 
when i switch the POW ON or OFF, then i must wait the telemetric period before i see power data.
Can you trigger to send telemetric data when the powerstaus is changed, then we get immediate feedback.

LG
Reinhart

patbeirne, 2017-01-13:
Reinhard,

The immediate feedback come in a "stat" message. 

For example, if you send the command:
`cmnd/mylight/power1 on`
the Sonoff will reply immediately with
`stat/mylight/POWER1 ON`

The "tele" message are send periodically to inform you of slow-changing or long-term states.

Hogesyx, 2017-01-14:
Reinhartd,

I understand your use case, this is that I do.

Your automation bus or the receiving unit should be the one that determine when they need the data. So for example your usual teleperiod might be 300 seconds, but it is possible to code the automation bus to send out a MQTT message to inform the sonoff to change to whatever interval(for example i use 5s) upon a POWER ON, then upon receiving a POWER OFF, it set back to 300 seconds.

Maybe we could split the TelePeriod to PowerOffTelePeriod PowerOnTelePeriod, but I still think that behavior such as this should be manage by the automation bus rather than end point.

littlebilly, 2017-01-14:
@Hogesyx
I think this i the right intention of Reinhartds suggestion.
Nevertheless it would be nice if Theo could implemet this into the code.
Just trigger the "tele/pow1/.." regarding Power ON / OFF.
An other open point which in my opinion only could be solved by Theo is the the following:
If for example the VOLTAGE would fall suddenly from 220 to 110 you never will get an alarm via MQTT
in the teleperiod slot.
I might be wrong?
Hope this helps to get th idea behind and may be a solution by Theo.

Reinhartd, 2017-01-14:
@patbeirne

it is right, when i send the command  "cmnd/mylight/power1 on"  
 then Sonoff will reply immediately with
 stat/mylight/POWER1 ON, but i get no information of current, voltage or power. This information i get only in the next teleperiod!

I would like to see these data immediately:
`tele/pow1/CURRENT_POWER 53
tele/pow1/VOLTAGE 214
tele/pow1/CURRENT 0.247
`
LG

Hogesyx, 2017-01-14:
@littlebilly 
I know what he needs, but messing with random telemetry call kind of defeat the purpose of telemetry interval, which provides a more or less accurate data point. Maybe instead of pumping out as telemetry, the initial power data can be pushed out as STATE instead.