arendst/Sonoff-MQTT-OTA-Arduino #237: Feature Request: Activate Wifi manager access point or other states via the wall switch

gizmochief7, 2017-01-14:
Lets say the Sonoff is configured with USE_WALL_SWITCH enabled and it was installed behind a wall plate and I wanted to launch the Wifi manager access point, It would be really nice to toggle the switch some number of times to activate its Wifi manager or other states that button presses give you. Something like 10 flicks in 5 seconds to restart it for example. Also, if someone doesn't want that option then it could be configured as so in user_config.h.

citrusfizz, 2017-01-14:
What are you using on the plate for the box?
I was thinking of getting a blank plate and putting a momentary button on it that was connected to the push button on the sonoff.   that way i could turn on/off the light and activate other features just like what is built in the push button already.     

I'm curious to what you are doing

jaysuk, 2017-01-14:
This has also concerned me. Not sure I want a sonoff installed where I can't reach it, just incase something goes wrong

gizmochief7, 2017-01-15:
@citrusfizz Sorry I used a poor choice of words to describe the scenario. lets say I have an existing [light switch](http://www.homedepot.com/p/Leviton-15-Amp-Single-Pole-Toggle-Switch-Light-Almond-R56-01451-02T/100669880) installed in a stand [1-Gang box](http://www.homedepot.com/p/Carlon-1-Gang-22-cu-in-New-Work-Box-B122A-UPC/202546989) in the wall. I want to remove the power running through the wall switch completely and run it through the Sonoff only. Next, I plan on having the Sonoff USE_WALL_SWITCH enabled allowing for external switch control. I will then hook the sensor pin 14 and a ground to the wall switch. This will allow me to control the Sonoff with the wall switch and keep all states synced within HA. I'll probably add a 30-50kv transitive diode just for good measure in case lighting strikes haha.  
Putting aside how I physically plan on getting the Sonoff in the Gang box and other parallel wiring circuit concerns, once the Sonoff is in there its not easily accessible to button presses. You might say I could just just hit the webpage and configure it but in this case lets assume for what ever reason its not configured to the network and i don't want to hook it up first to another power source first just to configure it before actually installing it. 

So my use case is this. I plan on making a few of these and then giving them away as a gifts. It would be nice if they can just hook them up first where ever they will be installed and flip the switch a few times to configure it. 

I feel this isn't that big of a source code add either for its benefits since all the source is already there for the button presses.

gizmochief7, 2017-01-15:
@citrusfizz I forgot to mention why i didn't want to hook up to the existing button for this. For any reason the AC power is shut off and comes back on while I happen to be pushing that button, the device gets fried or worse. Also the pins are already accessible to me for this use case.