arendst/Sonoff-MQTT-OTA-Arduino #242: Sonoff with Amazon Echo

thommycgn, 2017-01-14:
Hi! 
I just got my Amazon Echo (german version) today and tried to control the sonoffs with it. 
I was updating the sonoff to the newest Firmware (sonoff.ino.bin, v3.2.2). 
Friendly Name is "Wohnzimmer" f.e. 
How to go on?

I started the Alexa-App. Under Smarthome I taped on Search Devices (Geräte suchen) and: nothing found. Of course Alexa and the Sonoffs are in the same Wifi-Network.

Did I forget anything? Can anybody give me an instruction how to make it run?

don-willingham, 2017-01-15:
That sounds about right. Double check that the // is removed from #define USE_WEMO_EMULATION (it defaults to disabled/commented out). You can browse ip-of-sonoff/setup.xml. Enable logging and check for UPnP: Response sent to another-ip:port. Good luck.

don-willingham, 2017-01-15:
I found a problem with my sonoff and Alexa. Try connecting it to an MQTT broker, like mosquitto. Maybe I shouldn't have just created https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/245, maybe we'll mark that as a duplicate.

altelch, 2017-01-15:
Yes, there is a Problem if MQTT is enabled but no server can be connected. Then UDP gets disconnected on every retry and UDP pull won't work -> no Wemo emulation.

arendst, 2017-01-16:
Should be fixed in version 3.2.3 just released

closed 2017-01-24 by arendst