arendst/Sonoff-MQTT-OTA-Arduino #243: Problem with power / light command - traced to invalid index

kiwironnie, 2017-01-14:
Have flashed the latest version.  All is mostly working well, including OTA upgrades, however power and light mosquitto_pub, MQTT commands return a list of commands (i.e. effectively invalid command).

I've narrowed this down to 'index' validation at line 1117 in sonoff.ino

if ((!strcmp(type,"POWER") || !strcmp(type,"LIGHT")) && (index > 0) && (index <= Maxdevice)) {

It appears that index = atoi(type+i); at earlier line 1092 is returning an index value that causes the problem

By inserting index = 1; after this line, mosquitto_pub power and light commands succeed, e.g. mosquitto_pub -h host -t cmnd/switch1/POWER -m 1 switches the relay and led on.

Haven't traced the logic of what type+i is doing, or should be.

arendst, 2017-01-15:
Are you sure you are using version 3.2.2 as your line numbers do not reflect the source. I have no problem switching relays with version 3.2.2

See also #228

kiwironnie, 2017-01-15:
Thanks for getting back Theo.

You're right.  Downloaded 3.2.0 by mistake.  Doh!

#define VERSION                0x03020000   // 3.2.0

No doubt the latest version is ok.  So will close the (non) issue!

Very much appreciate the excellent thought and effort that you've put into the application.

Cheers, Ron.

closed 2017-01-15 by kiwironnie