arendst/Sonoff-MQTT-OTA-Arduino #246: Sonoff doesn't save and restore last POWER state

Imi3est, 2017-01-15:
Hi,
(sorry for my english)

I try to use sonoff with theese settings:
#define SAVE_STATE             1            // [SaveState] Save changed power state to Flash (0 = disable, 1 = enable)
#define APP_POWERON_STATE      3            // [PowerOnState] Power On Relay state (0 = Off, 1 = On, 2 = Toggle Saved state, 3 = Saved state)

It should save the POWER state always when it changes and restore the last value after power loss. The problem is that it doesn't saves states when I change the state via mqtt or via web admin. 
I chacked "Flash write count" on web interface and it doesn't increasing at POWER state change.
(the relay switches when I change the state so the communication works)

Thanks!

jaysuk, 2017-01-15:
you may need spiffs enabled for this to work

arendst, 2017-01-15:
Works fine with me. No need for spiffs.

aferraz, 2017-01-15:
Hi,
 same issue for me:
#define APP_POWERON_STATE      3     
#define SAVE_STATE             1  
but after a restart or after a power failure it remains in OFF

Thanks
 Alessandro

aferraz, 2017-01-15:
Sorry, I make some tests. The problem seem bound only to a unit. I have 6 units and only 1 give me this problem. Could be hardware problem.

Alessandro

Imi3est, 2017-01-16:
Hi,
I found the problem, it was my mistake: My setup was this:
#define SAVE_DATA              120            // [SaveData] Save changed parameters to Flash (0 = disable, 1 - 3600 seconds)
I thought SAVEDATA could be between 1 and 3600. But saving of POWER state works only if SAVE_DATA is 1.

Thanks for arendst to pointing out that it should be 1.
Gr, Imre

closed 2017-01-16 by Imi3est