arendst/Sonoff-MQTT-OTA-Arduino #248: Idea: use environments in PlatformIO to differentiate hardware

wiktorschmidt, 2017-01-16:
Hey @arendst,

I thought you might find this interesting.

A pretty cool pattern:
https://bitbucket.org/xoseperez/espurna/src/190a540cd2963215d54f87b211643a76b4b68e62/code/platformio.ini?at=master&fileviewer=file-view-default#platformio.ini-37

Can be used to replace the a little obscure define method:
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/blob/master/sonoff/user_config.h#L12

Just an idea.

arendst, 2017-01-16:
Have a look in the experimental folder. Almost no setup needed as all modules are compiled in as one image...
Once uuploaded all config is done online.

wiktorschmidt, 2017-01-16:
Interesting. I'll take a look. Thanks!

wiktorschmidt, 2017-01-17:
Works great. It's missing a platformio.ini the folder though :)

closed 2017-01-21 by arendst