arendst/Sonoff-MQTT-OTA-Arduino #249: Home Assistant MQTT query for Tempature/Humidity

citrusfizz, 2017-01-16:
I'm setting up Home Assistant and I got it reporting the on/off state of my sonoff and am able to switch it from the web interface of home assistant. 
I added a temp sensor on GPIO14 and I am able to see the temp/humidity on the main page of the arduino.   I did notice that is publishes this temp to my MQTT server every now and then. but how do i make a publish to query for the temp/humidity?

looking in the commands under the wiki here doesn't have anything about DHT or TEMP and querying for status  doesn't return the temp/humidity.  

Any thoughts?

citrusfizz, 2017-01-17:
looking through the code, it doesn't look like there is a way to request temp/humidity info from MQTT. 
I supposed i could scrape it from the root page. but it'd be nice to be able to use a cmnd to request it.

mihalski, 2017-01-18:
Why not just have a MQTT sensor in Home Assistant? There is no need for scraping.

mihalski, 2017-01-18:
Why not just use a MQTT sensor like so?

```
sensor
  - platform: mqtt
    name: "Office Temp Sonoff"
    state_topic: "tele/sonoff_th_office/+/TEMPERATURE"
    qos: 1
    unit_of_measurement: "°C"
  - platform: mqtt
    name: "Office Humidity"
    state_topic: "tele/sonoff_th_office/+/HUMIDITY"
    qos: 1
    unit_of_measurement: "%"
```

Easy.

citrusfizz, 2017-01-20:
damn dog stole my DHT22 sensor .  been trying to find it to test if that works @mihalski 

i did try something close to that with no luck,  but i won't be able to test until i either get another DHT22 or find the one the dog took (hopefully didn't eat it HA!)

incedently  @arendst   i saw in the user_config.h that it supports DHT11, I tried one of those sensors and got no results.  have you tried a DHT11?   Pretty sure they are analog where as the DHT22 is not. 

Correct me if i'm wrong

diogos88, 2017-02-12:
You can request the status whit the command : 
`Status 0`

You can receive the data with the HA MQTT Sensor : 
```
- platform: mqtt
  state_topic: "tele/sonoff/TELEMETRY"
  name: "Temperature"
  qos: 0
  unit_of_measurement: "°C"
  value_template: '{{ value_json.DHT.Temperature }}'

- platform: mqtt
  state_topic: "tele/sonoff/TELEMETRY"
  name: "Humidity"
  qos: 0
  unit_of_measurement: "%"
  value_template: '{{ value_json.DHT.Humidity }}'
```