arendst/Sonoff-MQTT-OTA-Arduino #250: Future of "old" Sonoff Branch

altelch, 2017-01-17:
Hi Theo,

what are your plans regarding the current Sonoff-MQTT-OTA-Arduino branch in contrast to the experimental branch? When do you plan to switch over to the new Version and how/where should the other developers integrate their code? I'm doing a lot of stuff right now based on the "old" branch so it would be good to know where to integrate the code into.

Thanks,
               Heiko.

arendst, 2017-01-17:
Stick to the "old" branch for the moment.

I'm currently integrating your WS2812 code with my WS2812 code and prepare for Sonoff Led (Cold/Warm but someday RGB).

I'm implementing the WS2812 code in a seperate ```xdrv_ws2812.ino``` file.

I also plan to make the Wemo/Hue code available in a saperate file ```xdrv_wemo.ino``` .

So for the moment please update your Wemo/Hue code but leave the ws2812 code as is as I've already changed a lot around adding commmands like Color, Dimmer, Fade etc.

altelch, 2017-01-17:
Hi,

thanks for the quick response :-) Ok, no further changes to LED-code then. I'm waiting to see how you implemented the commands. For the HUE part I'd need a way to manipulate the LEDs or STRIP internally like switching the relais. The Sonoff LED also uses WS2812? I'd guess they use another GPIO to controll the strip. The NeoPixelBus uses a DMA controlled Pin what creates no problems with  Wifi functions.

Thanks again,

              Heiko.

arendst, 2017-01-17:
The sonoff leds are PWM controlled so they are not WS2812. Rudimentary code in experimental version but I still haven't received my Sonoff Led so I can't test.

I'm currently struggling with intermittent led control with both DMA and pin14 configurations...

altelch, 2017-01-17:
Hi,

ok, looked into your experimental code. To have WS2812 we need to have a switch for the led config. Then I could use your setSColor function and try to emulate the same functions on WS2812. On the other hand I'd like to have individual controll over the LEDs in the strip what's a nice feature of the WS2812. So I think of a Sonoff-LED compatible mode and an extended Mode. But one step after the other. I'll wait what you come up with and then go on.

Greetings,

              Heiko.

jaysuk, 2017-01-17:
I can confirm that the sonoff LED works with the current experimental code.

arendst, 2017-01-18:
I will focus on the experimental code to get it production ready ASAP.

closed 2017-01-21 by arendst