arendst/Sonoff-MQTT-OTA-Arduino #252: user config update

jesusvallejo, 2017-01-18:
is there any way to update the userconfig without usin the usb com or the arduino ota (this is because my arduino ide cant find the comn port and i cant find any way to solve it). if there isnt it would be awesome an implimentation on the web update zone to update not only firmware but also user config. 

thank you

pd : ive triyed every thing , updating the ide , updating python , changing the router....

don-willingham, 2017-01-20:
I use "Export compiled binary" from the Sketch menu in the IDE. I then upload using the web interface, or using https://github.com/espressif/esptool from either github or from my Linux distribution.

arendst, 2017-01-21:
Many user_config parameters can be updated with online commands.

A future version will handle almost all options online.

closed 2017-01-24 by arendst